AsahiLinux/m1n1 #3: Makefile

below, 2021-01-15:
Could you include a makefile for this?

marcan, 2021-01-15:
I am currently building on amd64 Linux with an arch64-linux-gnu cross-compiler. This should build on aarch64 Linux with the native compiler (which should be the same thing). I imagine it should build on macOS with a gcc/binutils cross-compiler for that target, if you install one.

No libraries are used, m1n1 is pure free-standing code. The only compiler headers used are stdint.h and stdarg.h, which should be included in any basic toolchain.

below, 2021-01-15:
Sorry, I was under the impression this was meant to be a working sample for the clang toolchain (saw your post on Hackernews). 

While you are not using any libraries, the clang toolchain makes it hard to create a proper Mach-O binary without linking libSystem. I had thought you might have found an easy way to produce a minimal Mach-O executable

Shizmob, 2021-01-15:
The build system does create a Mach-O file natively through the toolchain, but instead builds one in the linker script, `m1n1.ld`, manually. AFAICS this is also a major hurdle to supporting lld or ld64, as I don't believe they support linker scripts.

As an aside @marcan: does it make sense to change the architecture to `aarch64-none-elf` here? m1n1 is bare-metal code, not a Linux user-space application, after all.

marcan, 2021-01-16:
The reason for using a userspace target is to use the same target as the OS we're going to run on, as it would be silly to ask even native users to install another cross-compiler. Usually whatever details that bit of the arch triplet implies can be changed with compiler flags anyway, so I'd like to keep that "lie" unless it actually causes a problem.

baryluk, 2021-02-05:
BTW. if you run on aarch64 Linux, it is quite easy to just change the first line of makefile or override this variable on command line to use compiler without prefix. Or install "cross-compiler" anyway. This is usually what I do, i.e. on Debian `apt install crossbuild-essential-arm64` will pull proper prefixed gcc and g++ packages, and they can be used on arm64 too.

marcan, 2021-05-22:
@baryluk At least on Gentoo the native compiler is available through the native arch prefix too, so I'm kind of surprised that on Debian you need a separate target for that. Maybe it's just about symlinks?

On Gentoo the chains go:

* /usr/bin/gcc -> /usr/x86_64-pc-linux-gnu/gcc-bin/11.1.0/gcc -> /usr/x86_64-pc-linux-gnu/gcc-bin/11.1.0/x86_64-pc-linux-gnu-gcc
* /usr/bin/x86_64-pc-linux-gnu-gcc -> /usr/x86_64-pc-linux-gnu/gcc-bin/11.1.0/x86_64-pc-linux-gnu-gcc

So as you can see they both end up in the same place.

Anyway, I think there isn't anything else to take care of here, so let's close :)

closed 2021-05-22 by marcan