AsahiLinux/m1n1 #14: Question： how does this code get control of M1?

leeehui, 2021-01-17:
I wonder how this code is loaded and start running from the very beginning. 
I am new here, and have too much info from [YouTube video](https://www.youtube.com/watch?v=d5s9fYfvzmY) and irc chat history. So quite appreciate if anyone can show more details to get this code run.
thanks a lot!

closed 2021-01-21 by leeehui

baryluk, 2021-02-05:
For reference: Things are explained in wiki in docs repo - for example - https://github.com/AsahiLinux/docs/wiki/Developer-Quickstart

leeehui, 2021-02-05:
@baryluk thanks! 
after two weeks of exploration, I have got m1n1 run.
actually may be a ”usb3 superspeed dual type-c cable” is the last thing I need to enable the serial console. usb type-c cable shipped  with macbook **can only reboot** another macbook via  [macvdmtool](https://github.com/AsahiLinux/macvdmtool/)