AsahiLinux/m1n1 #37: No more output after "Booting kernel at ..."

citruz, 2021-03-14:
First of all, thank you all for the effort put into this project!

I am trying to get Linux to boot and everything works up until m1n1 jumps into the kernel. There is no more output after that.
I followed the instructions [here](https://github.com/AsahiLinux/docs/wiki/Developer-Quickstart) and [here](https://github.com/AsahiLinux/docs/wiki/SW%3ALinux#building-linux) step by step and there were no errors whatsoever. The m1n1 log looks fine to me up to the last message `Booting kernel at 0x811200000 with fdt at 0x808d60000`.
Also tried embedding the kernel directly in m1n1 with exactly the same result.

My setup:
- Build machine: Debian 10 Arm VM running on M1
- Host: M1 Mac Mini connected to a M1 MacBook Air running `macvdmtool`

I am happy to debug it myself but I am not sure where to start. Any help is appreciated.

Full log:
```
python linux.py -b 'earlycon console=ttySAC0,1500000 console=tty0 debug' ../build/Image.gz ../build/apple-j274.dtb
Base at: 0x804690000
FB at: 0x9e0df8000
Setting boot args: "earlycon console=ttySAC0,1500000 console=tty0 debug"
.
Loading 1992012 bytes to 0x810e6c040..0x81105258c...
....................................................................................................................................................................................................................................................
Loading DTB to 0x8110525c0...
Kernel_base: 0x811200000
TTY> Starting secondary CPUs...
TTY> Starting CPU 1 (0:1)... OK
TTY>   Stack base: 0x804744040
TTY>   MPIDR: 0x80000001
TTY>   CPU: M1 Icestorm
TTY>   Index: 1 (table: 0x80475c080)
TTY>
TTY> Starting CPU 2 (0:2)... OK
TTY>   Stack base: 0x804748040
TTY>   MPIDR: 0x80000002
TTY>   CPU: M1 Icestorm
TTY>   Index: 2 (table: 0x80475c0c0)
TTY>
TTY> Starting CPU 3 (0:3)... OK
TTY>   Stack base: 0x80474c040
TTY>   MPIDR: 0x80000003
TTY>   CPU: M1 Icestorm
TTY>   Index: 3 (table: 0x80475c100)
TTY>
TTY> Starting CPU 4 (1:0)... OK
TTY>   Stack base: 0x804750040
TTY>   MPIDR: 0x80010100
TTY>   CPU: M1 Firestorm
TTY>   Index: 4 (table: 0x80475c140)
TTY>
TTY> Starting CPU 5 (1:1)... OK
TTY>   Stack base: 0x804754040
TTY>   MPIDR: 0x80010101
TTY>   CPU: M1 Firestorm
TTY>   Index: 5 (table: 0x80475c180)
TTY>
TTY> Starting CPU 6 (1:2)... OK
TTY>   Stack base: 0x804758040
TTY>   MPIDR: 0x80010102
TTY>   CPU: M1 Firestorm
TTY>   Index: 6 (table: 0x80475c1c0)
TTY>
TTY> Starting CPU 7 (1:3)... OK
TTY>   Stack base: 0x80475c040
TTY>   MPIDR: 0x80010103
TTY>   CPU: M1 Firestorm
TTY>   Index: 7 (table: 0x80475c200)
TTY>
TTY> FDT: bootargs = 'earlycon console=ttySAC0,1500000 console=tty0 debug'
TTY> FDT: framebuffer@9e0df8000 base 0x9e0df8000 size 0x7e9000
TTY> ADT: 64 bytes of random seed available
TTY> FDT: KASLR seed initialized
TTY> FDT: Passing 64 bytes of random seed
TTY> FDT: DRAM at 0x800000000 size 0x200000000
TTY> FDT: Usable memory is 0x80134c000..0x9db5e0000 (0x1da294000)
TTY> FDT: CPU 0 MPIDR=0x0 release-addr=0x80475c050
TTY> FDT: CPU 1 MPIDR=0x1 release-addr=0x80475c090
TTY> FDT: CPU 2 MPIDR=0x2 release-addr=0x80475c0d0
TTY> FDT: CPU 3 MPIDR=0x3 release-addr=0x80475c110
TTY> FDT: CPU 4 MPIDR=0x10100 release-addr=0x80475c150
TTY> FDT: CPU 5 MPIDR=0x10101 release-addr=0x80475c190
TTY> FDT: CPU 6 MPIDR=0x10102 release-addr=0x80475c1d0
TTY> FDT: CPU 7 MPIDR=0x10103 release-addr=0x80475c210
TTY> FDT prepared at 0x808d60000
Uncompressing gz ...
5627912
Decompress OK...
Ready to boot
DAIF: c0
--- TTY mode | Quit: CTRL+] | Menu: CTRL+T ---
MMU: shutting down...
MMU: shutdown successful, clearing caches
Booting kernel at 0x811200000 with fdt at 0x808d60000
```

marcan, 2021-03-14:
Ah, sorry, those instructions are obsolete. Please try the commit I just pushed to 'main' on the linux repo, and use the devicetree that's built with Linux (built as `arch/arm64/boot/dts/apple/t8103-j274.dtb`), not the one that comes with m1n1. There's been a lot of shuffling around with kernel patchsets and the devicetree, and the one in m1n1 no longer is in sync with anything.

marcan, 2021-03-14:
Just updated the m1n1 devicetree against the kernel one, so feel free to try that too :)

citruz, 2021-03-14:
Awesome, can confirm that both work. Thanks! :)

closed 2021-03-14 by citruz