AsahiLinux/m1n1 #91: tunable: unable to find ADT node /arm-io/apcie/pci-bridge1.

Zzzec, 2021-07-26:
`m1n1 base: 0x8031c8000
Setting boot args: "earlycon console=ttySAC0,1500000 console=tty0 debug"
Loading 6349721 bytes to 0x810448200..0x810a56599...
........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Loading DTB to 0x810a565c0...
Kernel_base: 0x810c00000
Loading 27650501 initramfs bytes to 0x830c00000...
................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
TTY> Starting secondary CPUs...
TTY> Starting CPU 1 (0:1)...   Started.
TTY> Starting CPU 2 (0:2)...   Started.
TTY> Starting CPU 3 (0:3)...   Started.
TTY> Starting CPU 4 (1:0)...   Started.
TTY> Starting CPU 5 (1:1)...   Started.
TTY> Starting CPU 6 (1:2)...   Started.
TTY> Starting CPU 7 (1:3)...   Started.
TTY> FDT: bootargs = 'earlycon console=ttySAC0,1500000 console=tty0 debug'
TTY> FDT: initrd at 0x830c00000 size 0x1a5e9c5
TTY> FDT: framebuffer@9df4e4000 base 0x9df4e4000 size 0xfa0000
TTY> ADT: 64 bytes of random seed available
TTY> FDT: KASLR seed initialized
TTY> FDT: Passing 64 bytes of random seed
TTY> FDT: DRAM at 0x800000000 size 0x200000000
TTY> FDT: Usable memory is 0x8019c4000..0x9d93e4000 (0x1d7a20000)
TTY> FDT: CPU 1 MPIDR=0x1 release-addr=0x803298110
TTY> FDT: CPU 2 MPIDR=0x2 release-addr=0x803298150
TTY> FDT: CPU 3 MPIDR=0x3 release-addr=0x803298190
TTY> FDT: CPU 4 MPIDR=0x10100 release-addr=0x8032981d0
TTY> FDT: CPU 5 MPIDR=0x10101 release-addr=0x803298210
TTY> FDT: CPU 6 MPIDR=0x10102 release-addr=0x803298250
TTY> FDT: CPU 7 MPIDR=0x10103 release-addr=0x803298290
TTY> FDT prepared at 0x808338000
Uncompressing gz ...
15026688
Decompress OK...
Ready to boot
DAIF: c0
TTY> tunable: unable to find ADT node /arm-io/apcie/pci-bridge1.
TTY> pcie: Error applying apcie-config-tunables for /arm-io/apcie/pci-bridge1
TTY> Preparing to boot kernel at 0x810c00000 with fdt at 0x808338000
`

I can boot into m1n1 successfully but when I try to boot the kernel, it always display 'unable to find ADT mode ...', then the screen stucks at a Apple icon. I think I have followed the steps in README. It looks like M1N1 have some trouble when parsing ADT?

Emantor, 2021-07-27:
What kind of M1 device are you running on?

marcan, 2021-07-27:
If you don't get a framebuffer console then something is wrong with your kernel; the apcie error doesn't have anything to do with the kernel not booting, it just means your hardware does not have that pcie device. It's not a fatal error.

Zzzec, 2021-07-28:
> If you don't get a framebuffer console then something is wrong with your kernel; the apcie error doesn't have anything to do with the kernel not booting, it just means your hardware does not have that pcie device. It's not a fatal error.

Thank you marcan, it's the problem when I build the kernel, my mistake.

Zzzec, 2021-07-28:
> What kind of M1 device are you running on?
> 
> Also, do you have a Debug TTY attached to the USB-C port? Upstream Linux only supports the debug TTY at the moment which requires extra hardware. If you want to use USB in gadget mode for this you'll need a patched kernel.

Macbook Air, the problem is that I have some wrong steps in building the kernel. Thanks for your help~

closed 2021-07-28 by Zzzec