AsahiLinux/m1n1 #95: A hunch about unkFloat in swap_submit_dcp()

kleinerm, 2021-08-14:
Just read the Asahi Linux status update, great work people! At your rate of progress, my own neuroscience OSS software might run well on Asahi Linux, long before it runs bearable on macOS 12 with M1 SoC's :)

Looking at the signature of swap_submit_dcp() and the actual DPC interface, i have a hunch that the unkFloat 64 Bit double in ...

https://github.com/AsahiLinux/m1n1/blob/c2c6da3df25c0605894244b4ea9387e882321efc/proxyclient/experiments/dcp.py#L279

... could be related to / an implementation of macOS Metal's MTLDrawable:presentAtTime: argument:
https://developer.apple.com/documentation/metal/mtldrawable/1470282-presentattime?language=objc

Iow. a double value presenting Mach host system time of when a swap should ideally occur? Assuming the DCP has a shared timebase with the ARM cpu, or some mapping happens? Would make sense, given that they offload other rather high-level things to the DCP.

Just a thought. Have a nice weekend
-mario

alyssarosenzweig, 2021-09-03:
This is a distinct possibility. Not for Metal specifically -- the DCP is uninvolved with Metal -- but presentation time in general. It's worth testing out the behaviour with nonzero floats passed.

marcan, 2021-09-04:
Note that there are already a number of timestamps in the swap request structure (that we currently ignore).

https://github.com/AsahiLinux/m1n1/blob/c2c6da3df25c0605894244b4ea9387e882321efc/proxyclient/m1n1/fw/dcp/ipc.py#L479

closed 2025-05-04 by svenpeter42