AsahiLinux/m1n1 #106: kernel image not aligned on 64K boundary (uboot via m1n1)

juliosueiras, 2021-09-14:
[image]

Hi, I am experimentation with uboot + nixos and wondering if this issue is related to m1n1 or uboot

juliosueiras, 2021-09-14:
Correction: the u-boot part is fine, the error occur during grub boot

marcan, 2021-09-22:
I'm confused; you say you're using grub, but that screenshot shows the EFI stub. Are you booting the kernel as an EFI application instead of as a native ARM64 kernel from GRUB?

We aren't really testing with GRUB yet, so various things might be broken... though this sounds like a bug in downstream components, not m1n1.

juliosueiras, 2021-09-23:
yea, thanks for response

closed 2021-09-23 by juliosueiras