AsahiLinux/m1n1 #107: Trouble in booting Linux under M1N1 Hypervisor.

Zzzec, 2021-09-18:
Here's the log when I directly boot the Linux. Although the Linux can boot up, the log on the host side shows like this.  

zec@ubuntu-zec:~/GitTools/m1n1/proxyclient/tools$ sudo M1N1DEVICE=/dev/ttyACM0 python3.9 linux.py -b 'earlycon console=SAC0,1500000 console=tty0 debug' asahi_Image.gz asahi.dtb initramfs-busybox.cpio.gz

TTY> CPU init... CPU: M1 Icestorm
TTY> 
TTY> boot_args at 0x808e08000
TTY>   revision:     2
TTY>   version:      2
TTY>   virt_base:    0xfffffe001b9c4000
TTY>   phys_base:    0x8019c4000
TTY>   mem_size:     0x1d7a20000
TTY>   top_of_kdata: 0x808e0c000
TTY>   video:
TTY>     base:       0x9df4e4000
TTY>     display:    0x1
TTY>     stride:     0x2800
TTY>     width:      2560
TTY>     height:     1600
TTY>     depth:      0x1001e
TTY>   machine_type: 0
TTY>   devtree:      0xfffffe001e508000
TTY>   devtree_size: 0x5c000
TTY>   cmdline:      
TTY>   boot_flags:   0x0
TTY>   mem_size_act: 0x200000000
TTY> 
TTY> 
TTY> 
TTY> m1n1 v63e8d0d-dirty
TTY> Copyright (C) 2021 The Asahi Linux Contributors
TTY> Licensed under the MIT license
TTY> 
TTY> Running in EL2
TTY> 
TTY> Heap base: 0x808e0c000
TTY> MMU: Initializing...
TTY> MMU: SCTLR_EL1: 30100180 -> 30901085
TTY> MMU: running with MMU and caches enabled!
TTY> fb init: 2560x1600 (30) [s=2560] @0x9df4e4000
TTY> fb console: max rows 46, max cols 64
TTY> fb: display logo
TTY> Device info:
TTY>   Model: MacBookAir10,1
TTY>   Target: J313
TTY>   Board-ID: 0x26
TTY>   Chip-ID: 0x8103
TTY> 
TTY> WDT registers @ 0x23d2b0000
TTY> WDT disabled
TTY> pmgr: initialized, 268 devices found.
TTY> Initialization complete.
TTY> Checking for payloads...
TTY> No more payloads at 0x8046b8000
TTY> No valid payload found
TTY> USB0: initialized at 0x808e0c030
TTY> USB1: initialized at 0x808e0c440
TTY> Running proxy...
m1n1 base: 0x8045c0000
Setting boot args: "earlycon console=ttySAC0,1500000 console=tty0 debug"
Loading 6364959 bytes to 0x811840200..0x811e5211f...
.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Loading DTB to 0x811e52140...
Kernel_base: 0x812000000
Loading 2024993 initramfs bytes to 0x832000000...
........................................................................................................................................................................................................................................................
TTY> Starting secondary CPUs...
TTY> Starting CPU 1 (0:1)...   Started.
TTY> Starting CPU 2 (0:2)...   Started.
TTY> Starting CPU 3 (0:3)...   Started.
[... 204 lines cut ...]
  File "/usr/local/bin/lib/python3.9/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<console>", line 1, in <module>
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/shell.py", line 129, in display
    val()
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/hv.py", line 1297, in start
    self.p.hv_start(self.entry, self.guest_base + self.bootargs_off)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 950, in hv_start
    return self.request(self.P_HV_START, entry, *args)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 624, in request
    return self._request(opcode, *args2, **kwargs)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 590, in _request
    reply = self.iface.proxyreq(req, reboot=reboot, no_reply=no_reply, pre_reply=None)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 357, in proxyreq
    return self.reply(self.REQ_PROXY)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 230, in reply
    reply += self.readfull(1)
  File "/home/zec/GitTools/m1n1/proxyclient/m1n1/proxy.py", line 167, in readfull
    block = self.dev.read(size - len(d))
  File "/usr/local/bin/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
>>> 

Some problem about serial things?

marcan, 2021-09-18:
Your direct Linux boot log is normal (the pcie errors are normal). If you don't see anything on the screen then there is some problem with the kernel itself, but that is not something you can debug without a serial cable or the hypervisor.

For the hypervisor, are you using matched versions of m1n1 and the proxyclient? Mixing and matching m1n1 and proxyclient versions is not supported in general. You need to chainload a m1n1 build that matches your proxyclient tree first. Basically, we only guarantee that chainload.py works with old versions; any other features may break at any time, and in particular the hypervisor breaks ABI very often, so you should always chainload before using it.

Zzzec, 2021-09-18:
> Your direct Linux boot log is normal (the pcie errors are normal). If you don't see anything on the screen then there is some problem with the kernel itself, but that is not something you can debug [...]

Thanks marcan, I boot the latest version of m1n1 on Macbook Air, but I still can not boot up the Linux under Hypervisor. 

The log on M1 looks like this. 
[image]

The log on host like this.
'''
Entering hypervisor shell. Type `start` to start the guest.
>>> start
Disabling other iodevs...
 - IODEV.UART
 - IODEV.FB
 - IODEV.USB1
 - IODEV.USB0_SEC
 - IODEV.USB1_SEC
Doing essential MMIO remaps...
Updating page tables...
PT[200000000:235200000] -> HW
PT[235200000:235204000] -> RESERVED VUART
PT[235204000:23b700420] -> HW
PT[23b700420:23b700424] -> RESERVED PMGR HACK
PT[23b700424:23b754000] -> HW
PT[23b754000:23b754010] -> RESERVED CPU_START
PT[23b754010:23d280088] -> HW
PT[23d280088:23d28008c] -> RESERVED PMGR HACK
PT[23d28008c:23d280098] -> HW
PT[23d280098:23d28009c] -> RESERVED PMGR HACK
PT[23d28009c:700000000] -> HW
Uploading ADT (0x51110 bytes)...
Improving logo...
Shutting down framebuffer...
Enabling SPRR...
Enabling GXF...
Jumping to entrypoint at 0x819450800
[cpu0] Pass: msr OSLAR_EL1, x0 = 0 (OK) (OSLAR_EL1)
[cpu0] Pass: mrs x0, HID5_EL1 = 2082df50e700df14 (HID5_EL1)
[cpu0] Pass: msr HID5_EL1, x0 = 2082df50e700df14 (OK) (HID5_EL1)
[cpu0] Pass: mrs x0, EHID9_EL1 = 600000811 (EHID9_EL1)
[cpu0] Pass: msr EHID9_EL1, x0 = 600000811 (OK) (EHID9_EL1)
[cpu0] Pass: mrs x0, EHID10_EL1 = 3000528002788 (EHID10_EL1)
[cpu0] Pass: msr EHID10_EL1, x0 = 3000528002788 (OK) (EHID10_EL1)
[cpu0] Pass: mrs x0, EHID20_EL1 = 618100 (EHID20_EL1)
[cpu0] Pass: msr EHID20_EL1, x0 = 618100 (OK) (EHID20_EL1)
[cpu0] Pass: mrs x0, EHID20_EL1 = 618100 (EHID20_EL1)
[cpu0] Pass: msr EHID20_EL1, x0 = 618100 (OK) (EHID20_EL1)
[cpu0] Pass: mrs x0, EHID20_EL1 = 618100 (EHID20_EL1)
[cpu0] Pass: msr EHID20_EL1, x0 = 618100 (OK) (EHID20_EL1)
[cpu0] Pass: mrs x1, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu0] Skip: msr CYC_OVRD_EL1, x1 = 2000000
[cpu0] Pass: mrs x1, ACC_CFG_EL1 = d (ACC_CFG_EL1)
[cpu0] Skip: msr ACC_CFG_EL1, x1 = d
CPUSTART W 23b754000+4:32 = 0x2
CPUSTART W 23b754000+8:32 = 0x2
[cpu0] Starting guest secondary 0:1
[cpu0]  CPU #1: RVBAR = 0x819450000
TTY> HV: Initializing secondary 1
TTY> HV: Entering guest secondary 1 at 0x819450000
[cpu1] Pass: msr OSLAR_EL1, x0 = 0 (OK) (OSLAR_EL1)
[cpu1] Pass: mrs x0, HID5_EL1 = 2082df50e700df14 (HID5_EL1)
...
[cpu1] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu3] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu1] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu3] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu3] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu4] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
...
'''
It seems like a dead loop. But the screen of M1 stucks like the picture.

By the way, it seems took more time in booting Linux directly when I use the latest version of m1n1.

And I want to ask that is this part of direct Linux boot log normal?

TTY> Preparing to boot kernel at 0x812000000 with fdt at 0x80a070000
--- TTY mode | Quit: CTRL+] | Menu: CTRL+T ---
Preparing to run next stage at 0x812000000...
Exception in thread rx:
Traceback (most recent call last):
File "/usr/local/bin/lib/python3.9/threading.py", line 973, in _bootstrap_inner
--- Exit TTY mode ---
self.run()
File "/usr/local/bin/lib/python3.9/threading.py", line 910, in run
self._target(*self._args, **self._kwargs)
File "/usr/local/bin/lib/python3.9/site-packages/serial/tools/miniterm.py", line 499, in reader
data = self.serial.read(self.serial.in_waiting or 1)
File "/usr/local/bin/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Because I don't see this part of log in https://github.com/AsahiLinux/docs/wiki/Developer-Quickstart#boot-a-linux-kernel.

marcan, 2021-09-18:
Yes, it is normal. The Quickstart log was taken via serial, not USB. We do not have console continuity between m1n1 and Linux for USB. If you do a direct Linux boot over USB, you will always lose the connection (and then it's up to Linux to give you gadget mode or whatever).

For the HV log, have you opened the guest UART terminal? You should have a /dev/ttyACM2 or similar that you can open with picocom. That is a virtual serial port that will give you the same continuous log from m1n1 to Linux as you saw on the quickstart guide (as long as you have the earlycon enabled on Linux - you'll have to put that argument into the devicetree since payload mode does not let you specify boot args).

Zzzec, 2021-09-18:
I think the kernel and dtb file are ok, because I can direct boot using same files, when I try to boot the same kernel under hypervisor, the screen always strucks like picture.

marcan, 2021-09-18:
Please paste the full log with the cpustart stuff. It looks like there is some issue there, but you trimmed that part of the log.

marcan, 2021-09-18:
Actually, I think I know what the problem is. This is a regression I introduced with the new cpuidle code. It's a performance problem but there is also something else.

marcan, 2021-09-18:
Should be fixed in the latest main (make sure to update the m1n1 build and chainload it). It was just a severe performance issue introduced by the new smp code, which causes m1n1 to time out initializing CPUs.

The hypervisor should work correctly now. Make sure to open the secondary serial device to get the guest UART log; if Linux still isn't booting or giving any output, then that's an issue with the kernel (either an early failure or you don't have earlycon enabled).

closed 2021-09-18 by marcan

Zzzec, 2021-09-18:
Here is the host side log of booting under hypervisor.

zec@ubuntu-zec:~/GitTools/m1n1/proxyclient/tools$ sudo M1N1DEVICE=/dev/ttyACM1 python3.9 run_guest.py -S m1n1-payload.macho 
TTY> CPU init... CPU: M1 Icestorm
TTY> 
TTY> boot_args at 0x807fd0000
TTY>   revision:     2
TTY>   version:      2
TTY>   virt_base:    0xfffffe000b9c4000
TTY>   phys_base:    0x8019c4000
TTY>   mem_size:     0x1d7a20000
TTY>   top_of_kdata: 0x807fd4000
TTY>   video:
TTY>     base:       0x9df4e4000
TTY>     display:    0x1
TTY>     stride:     0x2800
TTY>     width:      2560
TTY>     height:     1600
TTY>     depth:      30bpp
TTY>     density:    1
TTY>   machine_type: 0
TTY>   devtree:      0xfffffe000d570000
TTY>   devtree_size: 0x5c000
TTY>   cmdline:      
TTY>   boot_flags:   0x0
TTY>   mem_size_act: 0x200000000
TTY> 
TTY> 
TTY> 
TTY> m1n1 vb16bad6
TTY> Copyright (C) 2021 The Asahi Linux Contributors
TTY> Licensed under the MIT license
TTY> 
TTY> Running in EL2
TTY> 
TTY> Heap base: 0x807fd4000
TTY> MMU: Initializing...
TTY> MMU: SCTLR_EL1: 30100180 -> 30901085
TTY> MMU: running with MMU and caches enabled!
TTY> fb init: 2560x1600 (30) [s=2560] @0x9df4e4000
TTY> fb console: max rows 46, max cols 64
TTY> fb: display logo
TTY> Device info:
TTY>   Model: MacBookAir10,1
TTY>   Target: J313
TTY>   Board-ID: 0x26
TTY>   Chip-ID: 0x8103
TTY> 
TTY> AIC registers @ 0x23b100000
TTY> WDT registers @ 0x23d2b0000
TTY> WDT disabled
TTY> pmgr: initialized, 268 devices found.
TTY> Initialization complete.
TTY> Checking for payloads...
TTY> No more payloads at 0x803880000
TTY> No valid payload found
TTY> USB0: initialized at 0x807fd4030
TTY> USB1: initialized at 0x807fd4440
TTY> Running proxy...
Fetching ADT (0x0005C000 bytes)...
Disable iodev IODEV.USB1
Initializing hypervisor over iodev IODEV.USB0
TTY> Starting secondary CPUs...
TTY> Starting CPU 1 (0:1)...   Started.
TTY> Starting CPU 2 (0:2)...   Started.
TTY> Starting CPU 3 (0:3)...   Started.
TTY> Starting CPU 4 (1:0)...   Started.
TTY> Starting CPU 5 (1:1)...   Started.
TTY> Starting CPU 6 (1:2)...   Started.
TTY> Starting CPU 7 (1:3)...   Started.
[... 267 lines cut ...]
[cpu3] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu1] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu2] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu3] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu1] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu1] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu3] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu2] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu1] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu3] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu1] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu2] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu3] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu1] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu1] Skip: msr CYC_OVRD_EL1, x0 = 3000000
[cpu2] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Skip: msr CYC_OVRD_EL1, x0 = 2000000
[cpu1] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
[cpu2] Pass: mrs x0, CYC_OVRD_EL1 = 2000000 (CYC_OVRD_EL1)
...
It's a dead loop

> Please paste the full log with the cpustart stuff. It looks like there is some issue there, but you trimmed that part of the log.

marcan, 2021-09-18:
> It's a dead loop

It's not; it's a live loop that repeats until Linux boots its secondary cores. It's working as intended. The problem was just that all the spam and having to be handled by the Python was making everything so slow that some cores were timing out during initialization.

marcan, 2021-09-18:
Also, I just noticed a different regression that breaks booting kernels directly (but not under the hypervisor), so please try the direct boot with the latest version and see if that fixes it for you.

Zzzec, 2021-09-19:
> Also, I just noticed a different regression that breaks booting kernels directly (but not under the hypervisor), so please try the direct boot with the latest version and see if that fixes it for you.

It works! Not only booting under hypervisor, and direct boot Linux. Thank you marcan!

Zzzec, 2021-11-23:
Hi marcan, I use the latest version of m1n1 but the hypervisor can not work again.

Zzzec, 2021-11-23:
> Also, I just noticed a different regression that breaks booting kernels directly (but not under the hypervisor), so please try the direct boot with the latest version and see if that fixes it for you.

Hi marcan, I use the latest version of m1n1 but the hypervisor can not work again.

Zzzec, 2021-11-23:
I can't boot Linux under hypervisor, I'm trying to see what happen.

marcan, 2021-11-23:
Make sure you're using a MacBook Air devicetree. We no longer allow mismatched devicetrees, and only the Mac Mini one is upstream at this point.

Zzzec, 2021-11-23:
> Make sure you're using a MacBook Air devicetree. We no longer allow mismatched devicetrees, and only the Mac Mini one is upstream at this point.

Thanks marcan, so with Mac mini devicetree I can only boot Linux directly on MacBook Air?
I want to ask that how to get the MacBook Air one. Thank you!

marcan, 2021-11-26:
Try this branch, which should have the devicetrees (and other extra things): https://github.com/asahilinux/linux/tree/t6000/bringup-work

commit c3ba77ce (2021-09-18): hv_exc: Handle SYS_IMP_APL_CYC_OVRD in the fast path

m1n1 now uses this for SMP, and due to wfi FIQ leakage from the HV
timer this gets spammed thousands of times per second. Handle it in
the HV directly.

Fixes: #107