AsahiLinux/m1n1 #111: Can't use keyboard when running Linux under hypervisor.

Zzzec, 2021-09-23:
I can use keyboard by using https://github.com/AsahiLinux/linux/tree/dart/dev this kernel when direct boot.
It doesn't work when I run the same kernel under hypervisor, although the log on Linux shows that it can detect the USB device.

svenpeter42, 2021-09-24:
Can you show the dmesg output for both cases?

And is there a specific reason for using that branch? DART has been merged upstream at this point and the code there is probably outdated.

Zzzec, 2021-09-27:
> Can you show the dmesg output for both cases?
> 
> And is there a specific reason for using that branch? DART has been merged upstream at this point and the code there is probably outdated.

Great, I don't know it has been merged upstream. Because I need to use keyboard to do some experiment on M1, I follow the steps in https://github.com/AsahiLinux/docs/wiki/SW%3ALinux#linux-usb-keyboard. It says that I need to use that /dart/dev branch.

?// I use the v5.15-rc3 and turn on the CONFIG_APPLE_DART=y, but the keyboard can't be used as using /dart/dev branch.

Now I can see the output by picocom another ttyACM using latest kernel version v5.15-rc3. Thanks.

closed 2021-09-28 by Zzzec