AsahiLinux/m1n1 #120: FDT: DT CPU 2 MPIDR mismatch 0x2 != 0x10100

rapidddenis, 2021-11-09:
Trying to boot Asahi Linux.

sudo python3.9 m1n1/proxyclient/tools/linux.py -b 'earlycon console=ttySAC0,1500000 console=tty0 debug'   Image.gz t8103-j274.dtb initrd.gz
m1n1 base: 0x1000461c000
Setting boot args: "earlycon console=ttySAC0,1500000 console=tty0 debug"
Loading 6336201 bytes to 0x10013a70200..0x1001407b0c9...
...........................................................................................................................................................................................................
Loading DTB to 0x1001407b100...
Kernel_base: 0x10014200000
Loading 27649945 initramfs bytes to 0x10034200000...
...........................................................................................................................................................................................................
TTY> Starting secondary CPUs...
TTY> Starting CPU 1 (0:1)...   Started.
TTY> Starting CPU 2 (1:0)...   Started.
TTY> Starting CPU 3 (1:1)...   Started.
TTY> Starting CPU 4 (1:2)...   Started.
TTY> Starting CPU 5 (1:3)...   Started.
TTY> Starting CPU 6 (2:0)...   Started.
TTY> Starting CPU 7 (2:1)...   Started.
TTY> Starting CPU 8 (2:2)...   Started.
TTY> Starting CPU 9 (2:3)...   Started.
TTY> FDT: bootargs = 'earlycon console=ttySAC0,1500000 console=tty0 debug'
TTY> FDT: initrd at 0x10034200000 size 0x1a5e799
TTY> FDT: framebuffer@107d1b10000 base 0x107d1b10000 size 0x1d73c00
TTY> ADT: 64 bytes of random seed available
TTY> FDT: KASLR seed initialized
TTY> FDT: Passing 64 bytes of random seed
TTY> FDT: DRAM at 0x10000000000 size 0x800000000
TTY> FDT: Usable memory is 0x10001e28000..0x107cb154000 (0x7c932c000)
TTY> FDT: CPU 1 MPIDR=0x1 release-addr=0x10004a8c050
TTY> FDT: DT CPU 2 MPIDR mismatch: 0x2 != 0x10100
DT prepare failed

jannau, 2021-11-09:
`t8103-j274.dtb` is the dtb for the M1 and won't work for M1 Pro/Max.

See [t6000-j314s.dts](https://github.com/AsahiLinux/linux/blob/t6000/bringup-work/arch/arm64/boot/dts/apple/t6000-j314s.dts) from `t6000/bringup` for a dts for M1 Pro/Max. Should work on all 2021 Macbook Pro models with linux.py for now.

rapidddenis, 2021-11-10:
It works!

closed 2021-11-10 by rapidddenis