AsahiLinux/m1n1 #128: Potential need for adjustment to Initial Kernel DCP Support (12.1)

Daasin, 2021-11-24:
Would the DCP firmware for 12.1 which is getting ready to release and in public betas (RC) now come under initial support from Asahi given the timing? 🤔

"Initial kernel DCP support expected for the firmware released with macOS 12 “Monterey” (which is currently in public beta); perhaps 12.0 **_or a newer point release, depending on the timing_**.'

Daasin, 2021-11-25:
Any updates on this by any chance? 🤔

marcan, 2021-11-25:
This isn't decided yet and probably won't be decided until the DCP driver gets a look-over in the near future. We also have an issue where 12.0.1 broke HDMI iBoot framebuffer output on the Mac Mini, which will need some extra investigation to see if we need to wait for Apple to fix it or we can do something about it.

Daasin, 2021-11-25:
Ah heard other similar issues are popping up too and then saw a patch note saying they'd support more GPU's

jannau, 2022-02-19:
I think this issue can be closed. The adapted dcp driver works with iboot from macos 12.1. As far as I can tell, there have been no changes for the core dcp functionality through 12.0 to 12.2.

Daasin, 2022-02-19:
> I think this issue can be closed. The adapted dcp driver works with iboot from macos 12.1. ***As far as I can tell, there have been no changes for the core dcp functionality through 12.0 to 12.2.*** 

Do you know if that would that also apply to the 12.3 beta preview by any chance? 🤔

marcan, 2022-02-19:
We're targeting 12.1 for the initial release; there's no particular rush to worry about later versions. DCP firmware is OS-tied, so it doesn't matter if you run a later macOS version. That's just the minimum you need for system firmware.

Daasin, 2022-02-19:
But if so on the 12.3 question, I shall close with no further comments ☺️   
 
Appreciate the additional info! 🤟🏼

closed 2022-02-19 by Daasin