AsahiLinux/m1n1 #135: kmutil returns: Error: Invalid Mach-O boot properties

hogliux, 2021-12-16:
I've tried following the QuickStart instructions to the dot on a MacBook Pro with M1 Pro. The mac came with macOS monteray pre-installed.

However, I hit a snag when trying to use `kmutil`:

```
-bash-3.2# kmutil configure-boot -c m1n1.macho -C -v /Volumes/Linux
By setting a custom boot object, you will be putting your system into Permissive Security.

Are you sure you want to do this? (enter y or n) y
found valid Mach-O binary, deriving Mach-O boot properties...
Error Domain=KMErrorDomain Code=71 "Fileset Error: Invalid Mach-O boot properties" UserInfo={NSLocalizedDescription=Fileset Error: Invalid Mach-O boot properties}
```

Any ideas?

NightRaven055, 2021-12-16:
I have the same issue

marcan, 2021-12-17:
Please run this from macOS or 1TR:

`curl -L mrcn.st/alxsh | sh`

Type in your password if prompted, press enter when prompted, then `q` at the action menu and paste the full output.

Also, please upload your `m1n1.macho` somewhere so we can take a look.

jannau, 2021-12-17:
This error is reported when then the specified volume is not the default startup disk. I'm not sure if that is the only context this error is reported in.
The startup disk can be selected from macos system setting or from 1tr. Select first options and then `Startup Disk` in the apple menu.

hogliux, 2021-12-17:
Thank you everybody for your help.

@marcan
This is the ouput of `curl -L mrcn.st/alxsh | sh`. It looks like the installer thinks that my Linux volume is the recovery volume.

```
-bash-3.2# curl -L mrcn.st/alxsh | sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   169  100   169    0     0   3643      0 --:--:-- --:--:-- --:--:--  4225
100   169  100   169    0     0   1186      0 --:--:-- --:--:-- --:--:--  1186
100   169  100   169    0     0   1050      0 --:--:-- --:--:-- --:--:--  1050
100   557  100   557    0     0   2159      0 --:--:-- --:--:-- --:--:--  2159

Bootstrapping installer:
  Downloading...
  Extracting...
  Initializing...

Welcome to the Asahi Linux installer!

This installer is in a pre-alpha state, and will only do basic
bootloader set-up for you. It is only intended for developers
who wish to help with Linux bring-up at this point.

Please make sure you are familiar with our documentation at:
  https://alx.sh/w

Press enter to continue.

Collecting system information...
System information:
  Product name: MacBook Pro (14-inch, 2021)
  SoC: Apple M1 Pro
  Device class: j314sap
  Product type: MacBookPro18,3
  Board ID: 0x8
  Chip ID: 0x6000
  System firmware: iBoot-7429.61.2
  Boot UUID: EC5DCF3D-63DB-40D0-BF44-8A15B10DB859
  Boot VGID: 23638A67-F527-4467-8F06-7E200F38E5AE
  Default boot VGID: 23638A67-F527-4467-8F06-7E200F38E5AE
  Boot mode: one true recoveryOS
  OS version: 12.1 (21C52)
  System rOS version: 12.1 (21C52)
  No Fallback rOS
  Login user: None

Collecting partition information...
  System disk: disk0

Collecting OS information...

Partitions in system disk (disk0):
  1: APFS [Macintosh HD] (768.00 GB, 6 volumes)
    OS: [  ] [Macintosh HD] macOS v12.1 [disk4s1, 24EB9A2B-CCCA-4DC8-9E91-EB26A65CEB97]
  2: APFS [Linux] (80.00 GB, 6 volumes)
    OS: [R*] [Linux] macOS v12.1 [disk3s3, 23638A67-F527-4467-8F06-7E200F38E5AE]
  3: Microsoft Basic Data (999.29 MB)
  4: Microsoft Basic Data (145.66 GB)
  5: APFS (System Recovery) (5.37 GB, 2 volumes)
    OS: [  ] recoveryOS v12.1 [Primary recoveryOS]

  [B ] = Booted OS, [R ] = Booted recovery, [? ] = Unknown
  [ *] = Default boot volume

No actions available on this system.
```

You can download the m1n1.macho binary [here](https://gist.github.com/hogliux/34edadf2474fd7da9ce5aab33da7982f#file-m1n1-macho). I also tried corellium's binary at https://downloads.corellium.info/linux.macho, but I get the same error.

> This error is reported when then the specified volume is not the default startup disk.

I tried changing the startup disk to Linux but I still get the same error when executing

marcan, 2021-12-17:
> This error is reported when then the specified volume is not the default startup disk. I'm not sure if that is the only context this error is reported in. The startup disk can be selected from macos system setting or from 1tr. Select first options and then `Startup Disk` in the apple menu.

That is not my experience. If you are not running the paired recovery, you should get a "Pairing" error.

marcan, 2021-12-17:
This is a kmutil change in 12.1 final (sigh, it wasn't in the betas...). We'll have to figure out what it wants.

hogliux, 2021-12-17:
OK no problem. Is there a way to download the kmutil beta from somewhere?

marcan, 2021-12-17:
We need to figure out what changed. The goal of this project is to make things work, not to work around problems.

marcan, 2021-12-17:
Looks like Apple added a new feature just for us. Try `kmutil configure-boot -c m1n1.macho -v /Volumes/Linux --raw --entry-point 34816 --lowest-virtual-address 16384`.

marcan, 2021-12-17:
Note that this will not work with payloads at the moment, due to the way the Mach-O file is constructed. But it will work for just m1n1.

closed 2021-12-17 by marcan

hogliux, 2021-12-17:
This works fine now. Thank you!

commit 9e63fe24 (2021-12-17): README.md: Update to mention m1n1.bin target

Fixes: #135