AsahiLinux/m1n1 #148: [help] question with the commands

Galorlx, 2022-01-17:
Any help with this "kmutil configure-boot -c m1n1.bin --raw --entry-point 2048 --lowest-virtual-address 0 -v <path to your OS volume>" where it says <path to your OS volume> is it my linux volume or my mac os volume and where would I put my linux volume if so

Galorlx, 2022-01-17:
If it doesn’t show fully
[image]

Galorlx, 2022-01-17:
If the comment doesn’t fully show 
[image]

jmranger, 2022-01-29:
That command has been integrated in the latest version of the Asahi installer.
See https://github.com/AsahiLinux/asahi-installer/commit/1009a195fa2b215d8eb741d41eb45b1c2bc3edaa or https://github.com/AsahiLinux/asahi-installer/blob/main/src/step2.sh

Galorlx, 2022-01-29:
> That command has been integrated in the latest version of the Asahi installer.
> See [AsahiLinux/asahi-installer@1009a19](https://github.com/AsahiLinux/asahi-installer/commit/1009a195fa2b215d8eb741d41eb45b1c2bc3edaa) or https://github.com/AsahiLinux/asahi-installer/blob/main/src/step2.sh

Is there documentation to fully install Linux instead of these mini things? (Thanks for this link)

jmranger, 2022-01-30:
While there's a lot working already, Asahi Linux is still at the "get things working" stage, not the "spit and polish" one. The installer is still considered "pre-alpha".
There is documentation in the wiki (https://github.com/AsahiLinux/docs/wiki/), but it's nowhere near what you get from your usual distro.
I'd encourage you to read the last progess reports (https://asahilinux.org/blog/) to get a better feel of the current status.
Given that this isn't an "issue" per se, I'd encourage you to close this ticket and continue the discussion on IRC (links on https://asahilinux.org/community/).

davide125, 2025-02-17:
https://asahilinux.org/docs/m1n1-User-Guide/ has details on how to setup m1n1 manually. For regular use we recommend just using the installer.

closed 2025-02-17 by davide125

ipatch, 2025-09-30:
> https://asahilinux.org/docs/m1n1-User-Guide/ has details on how to setup m1n1 manually. For regular use we recommend just using the installer.

link broken