AsahiLinux/m1n1 #155: python fails to load ADT since calculated base address ends up negative

svenpeter42, 2022-01-27:
I've only hit the following problem once with 12.1 firmware and couldn't reproduce it after a reboot now.

I've added some debug prints and m1n1 computes the ADT base address as `0x12c0000 - 0xffffffffff0a8000 + 0x8010a8000 = -0xfffffff7fcd40000`.
The same thing works in C because it overflows back to `0x8032c0000`.

I'm not sure if this is just a python issue and casting the result down to a u64 is good enough or if `0xffffffffff0a8000` is an unexpected `virt_base`. 

```
TTY> CPU init (MIDR: 0x611f0221)...
TTY>   CPU part: 0x22 rev: 0x11
TTY>   CPU: M1 Icestorm
TTY>
TTY> boot_args at 0x803b3c000
TTY>   revision:     2
TTY>   version:      2
TTY>   virt_base:    0xffffffffff0a8000
TTY>   phys_base:    0x8010a8000
TTY>   mem_size:     0x1df98c000
TTY>   top_of_kdata: 0x803b40000
TTY>   video:
TTY>     base:       0x9e3a5c000
TTY>     display:    0x1
TTY>     stride:     0xa00
TTY>     width:      640
TTY>     height:     1136
TTY>     depth:      32bpp
TTY>     density:    1
TTY>   machine_type: 0
TTY>   devtree:      0x12c0000
TTY>   devtree_size: 0x58000
TTY>   cmdline:
TTY>   boot_flags:   0x0
TTY>   mem_size_act: 0x200000000
TTY>
TTY>
TTY>
TTY> m1n1 v69ed3e4-dirty
TTY> Copyright The Asahi Linux Contributors
TTY> Licensed under the MIT license
TTY>
TTY> Running in EL2
TTY>
TTY> Device info:
TTY>   Model: Macmini9,1
TTY>   Target: J274
TTY>   Board-ID: 0x22
TTY>   Chip-ID: 0x8103
TTY>
TTY> Heap base: 0x803b40000
TTY> MCC: Initializing T8103 MCC...
TTY> MCC: Initialized T8103 MCC (8 channels)
TTY> MMU: Initializing...
TTY> MMU: Adding nGnRE mapping at 0x400000000 (0x80000000)
TTY> MMU: Adding nGnRE mapping at 0x480000000 (0x80000000)
TTY> MMU: Adding nGnRE mapping at 0x580000000 (0x80000000)
TTY> MMU: Adding nGnRE mapping at 0x600000000 (0x80000000)
TTY> MMU: Adding nGnRE mapping at 0x6a0000000 (0x20000000)
TTY> MMU: Adding nGnRE mapping at 0x6c0000000 (0x40000000)
TTY> MMU: Adding nGnRE mapping at 0x700000000 (0x100000000)
TTY> MMU: RAM base: 0x800000000
TTY> MMU: Top of normal RAM: 0x9e0a34000
TTY> MMU: Unmapping TZ0 region at 0x9e0a3c000..0x9e275c000
TTY> MMU: Unmapping TZ1 region at 0x9e0a34000..0x9e0a3c000
TTY> MMU: Unmapping TZ3 region at 0x9e6e64000..0x9ffe64000
TTY> MMU: SCTLR_EL1: 30100180 -> 30901085
TTY> MMU: running with MMU and caches enabled!
TTY> display: Dummy framebuffer found, initializing display
TTY> dart: dart /arm-io/dart-dcp at 0x23130c000 is a t8020 (locked)
TTY> dart: dart /arm-io/dart-disp0 at 0x231304000 is a t8020
TTY> rtkit(dcp): booting with version 12
TTY> rtkit(dcp): unknown system endpoint 0x0a
TTY> rtkit(dcp): pre-allocated buffer (ep 0x1, dva 0xf00243000, phys 0x9e6263000)
TTY> EPIC: started interface 0 (disp0-service)
TTY> display: waited 2000 ms for display status
TTY> display: connected:0 timing_cnt:0 color_cnt:0
TTY> fb init: 640x1136 (32) [s=640] @(nil)
TTY> fb console: max rows 31, max cols 4
TTY> fb: display logo
TTY> AIC: Version 1 @ 0x23b100000
TTY> WDT registers @ 0x23d2b0000
TTY> WDT disabled
TTY> pmgr: Cleaning up device states...
TTY> pmgr: Enabling SIO_ADMA, parent of active device MCA1
TTY> pmgr: Enabling ANS2, parent of active device APCIE_ST
TTY> pmgr: Enabling DEBUG, parent of active device DEBUG_USB
TTY> pmgr: initialized, 268 devices found.
TTY> cpufreq: Initializing clusters
TTY> cpufreq: Switching cluster PCPU to P-State 7
TTY> Initialization complete.
TTY> Checking for payloads...
TTY> Devicetree compatible value: apple,j274
TTY> Unknown payload at 0x803438000 (magic: 3083702c)
TTY> No valid payload found
TTY> dart: dart /arm-io/dart-usb0 at 0x382f80000 is a t8020
TTY> USB0: initialized at 0x803f46030
TTY> dart: dart /arm-io/dart-usb1 at 0x502f80000 is a t8020
TTY> USB1: initialized at 0x803f46360
TTY> Running proxy...
readmem: 803b3c000 2e0
m1n1 base: 0x803378000
LOAD: _HDR 16384 bytes from 0 to 0
LOAD: TEXT 163840 bytes from 4000 to 4000
LOAD: RODA 32768 bytes from 2c000 to 2c000
LOAD: DATA 393216 bytes from 34000 to 34000
ZERO: 196608 bytes from 0x94000 to 0xc4000
LOAD: PYLD 0 bytes from 94000 to c4000
SKIP: 67108864 bytes from 0xc4000 to 0x40c4000
12c0000 ffffffffff0a8000 8010a8000
Fetching ADT (0x00058000 bytes)...
readmem: -fffffff7fcd40000 58000
Traceback (most recent call last):
  File "/Users/speter/asahi/git/m1n1/proxyclient/tools/chainload.py", line 37, in <module>
    sepfw_start, sepfw_length = u.adt["chosen"]["memory-map"].SEPFW
  File "/Users/speter/asahi/git/m1n1/proxyclient/m1n1/proxyutils.py", line 326, in __getitem__
    return self._adt[item]
  File "/Users/speter/.asdf/installs/python/3.10.1/lib/python3.10/functools.py", line 970, in __get__
    val = self.func(instance)
  File "/Users/speter/asahi/git/m1n1/proxyclient/m1n1/proxyutils.py", line 324, in _adt
    return adt.load_adt(self._utils.get_adt())
  File "/Users/speter/asahi/git/m1n1/proxyclient/m1n1/proxyutils.py", line 184, in get_adt
    self.adt_data = self.iface.readmem(adt_base, self.ba.devtree_size)
  File "/Users/speter/asahi/git/m1n1/proxyclient/m1n1/proxy.py", line 385, in readmem
    req = struct.pack("<QQ", addr, size)
struct.error: argument out of range
```

phire, 2022-02-03:
I've hit this same bug with 12.2 firmware.

Managed to reproduce it after six reboots. Masking the address to 64bits does seem to produce the correct results.

marcan, 2022-02-06:
Looks like this is an artifact of using a somewhat weird virt base when we boot in raw mode. I'll just mask it out for now, seems like this doesn't break anything else.

commit 0d35eb3d (2022-02-06): m1n1.proxyutils: Wrap ADT base to 64 bits

This works around a wraparound issue when the virt base ends up causing
the ADT address to go negative, when on 12.1+ using a dummy virt base.

Fixes: #155
closed 2022-02-06 by marcan