AsahiLinux/m1n1 #157: run_guest.py fails to boot linux when using a concatenated initrd

jmranger, 2022-01-28:
Tested with m1n1 88874936c66a
Reporting since this works fine when booting with linux.py. Issue is only in run_guest.py.

Use case: when installing a rootfs with the initrd flavor of Debian's installer [0], it's useful to be able to provide a second initrd containing the Broadcom firmwares.

Commands used:
```
cat DebianBookwormInstaller.20220125.initrd.gz WiFiFW-MacOS12.1.cpio.gz > merged.cpio.gz
cat m1n1/build/m1n1.macho <(echo 'boot-args=earlycon console=ttySAC0,1500000 console=tty0') linux/arch/arm64/boot/dts/apple/t8103-j313.dtb linux/arch/arm64/boot/Image.gz initrd >| /tmp/m1n1-linux.macho
M1N1DEVICE=/dev/ttyACM0 python3 m1n1/proxyclient/tools/run_guest.py /tmp/m1n1-linux.macho
```

With `DebianBookwormInstaller.20220125.initrd.gz` as `initrd`, boot is successful.
When `merged.cpio.gz`, linux fails to boot with the following error:
```
[    1.616933] /dev/root: Can't open blockdev
[    1.618751] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
```

Logs attached.

I would totally understand if you'd choose to close this as wontfix.

[0] https://d-i.debian.org/daily-images/arm64/20220125-02:20/netboot/debian-installer/arm64/initrd.gz

[bad.txt](https://github.com/AsahiLinux/m1n1/files/7962410/bad.txt)
[good.txt](https://github.com/AsahiLinux/m1n1/files/7962411/good.txt)

marcan, 2022-01-29:
Try gunzipping both initrds, concatenating them, and gzipping them up again.

Explanation: m1n1 gunzips the initrds in order to calculate their length, but doesn't attempt to concatenate the result, so the second one will probably override the first one.

jmranger, 2022-01-29:
Tested. That works. Thanks.
Feel free to close. As mentioned, I reported mostly because I was expecting (possibly incorrectly) that both loaders would support similar use-cases.

marcan, 2022-01-31:
This is still a nit that's worth fixing; it should be possible to tweak the payload code to append initrds instead of replacing them, at least if they're consecutive in the payload order.