AsahiLinux/m1n1 #158: Unknown payload at 0x(varies) (magic 30837271)

jmranger, 2022-01-29:
I believe there's a glitch in the support of new bin file format for MacOS 12.1 stubs. Tested with version 8887493, which is the one in the current version of the pre-alpha installer.

The bootlog now contains the title line in all cold boots.

I did rebuild that version locally, and installed it on both a 12.0 and a 12.1 stub. Issue only present on 12.1. Issue is not present when chainloading, on both 12.0 and 12.1.

The only impact I found is the line on the screen - everything seems to work. So don't hesitate to close wontfix.

marcan, 2022-01-31:
That's probably just because iBoot no longer initializes any memory after m1n1 itself, so it sees garbage as the (nonexistent) first payload. That means this is an installer issue; it should concatenate a dummy payload of zeroes at the end of the m1n1 it builds.

jmranger, 2022-01-31:
That makes sense. But I'm somewhat surprised that the magic number is always the same. The address at which it is found varies.
I believe the location in memory where m1n1 is loaded varies from boot to boot, but if you tell me which other address to compare with, I could check whether the offset is constant from one cold boot to another.

jmranger, 2022-02-01:
So the offset seems to be constant.
Captured by cold-booting, then running pcie_enable_devices.py (since retyping isn't fun.)

```
TTY> Heap base: 0x803c60000
TTY> Unknown payload at 0x803528000 (magic: 30837331)
TTY> USB0: initialized at 0x803c8c070
TTY> USB1: initialized at 0x803c8c560
```

```
TTY> Heap base: 0x804590000
TTY> Unknown payload at 0x803e58000 (magic: 30837331)
TTY> USB0: initialized at 0x8045bc070
TTY> USB1: initialized at 0x8045bc560
```

```
TTY> Heap base: 0x803f08000
TTY> Unknown payload at 0x8037d0000 (magic: 30837331)
TTY> USB0: initialized at 0x803f34070
TTY> USB1: initialized at 0x803f34560
```

marcan, 2022-02-01:
That's because the offset is randomized in memory (KASLR), but right after m1n1 in memory is the SEP firmware which is an img4 image. You're seeing the first few bytes of that, so it's not really random garbage.

closed 2022-02-01 by marcan

marcan, 2022-02-01:
Should be fixed in fbc15bd, please give it a go :)

jmranger, 2022-02-01:
Tested on 12.0 and 12.1 stubs, with both run_guest.py and linux.py. The warning is gone, and no side effects observed.
Thanks!

commit 5c88e2d6 (2022-02-01): payload: Consider an IMG4 blob as the end of payloads

When loading m1n1.bin, SEPFW follows m1n1 and thus will show up at
the payload offset if there are no payloads.

Fixes: #158
commit fbc15bd7 (2022-02-01): payload: Consider an IMG4 blob as the end of payloads

When loading m1n1.bin, SEPFW follows m1n1 and thus will show up at
the payload offset if there are no payloads.

Fixes: #158