AsahiLinux/m1n1 #159: Mac mini 12+ DCP initialization fails with some monitors

VinDuv, 2022-01-29:
I have two monitors: An iiyama ProLite E2473HDS (1080p) and a Philips 279P1 (4K). They do not work with m1n1 from a 12.1 Mac mini (connected over HDMI).
When m1n1 starts, the 1080p monitor wakes up from sleep (power LED turns blue), and a few seconds later displays “No signal” and goes back to sleep. m1n1 logs (including the DCP syslog messages) indicate nothing out of the ordinary; the display is correctly recognized at the correct resolution/refresh rate. In fact, the log is identical to the one produced when the mini is connected to a 1080p capture card.
The 4K monitor does not seem to even wake up.

I’ve messed with m1n1’s display init code (mostly with the 1080p monitor since it’s not my primary display) and found out the following:

- The monitor wakes up from sleep when `dcp_ib_swap_begin` is called.
- Calling `dcp_ib_set_mode` after `dcp_ib_swap_begin` does not change anything at that point.
- Calling `dcp_ib_get_hpd` in a loop after `dcp_ib_swap_begin` shows that `connected` goes back to `0` after 1-2 seconds, then goes back to `1` after ~400ms.
- Calling `dcp_ib_set_mode` after `connected` is back to `1`, then completing the swap (`dcp_ib_swap_set_layer`/`dcp_ib_swap_end`) makes the monitor work correctly; it will actually show m1n1 logo/console. It still takes a couple of seconds after completing the swap for the image to actually appear.
- When rebooting, the situation is messier; after the call to `dcp_ib_swap_begin`, the monitor will still disconnect/reconnect, but after 5-6 seconds, not 1-2 seconds. It is, however, not necessary to wait for the disconnection: calling `dcp_ib_set_mode` a couple of seconds after `dcp_ib_swap_begin` seems to prevent the disconnection, and the monitor works afterwards.

I managed to make both my monitors work by making [these changes](https://github.com/VinDuv/m1n1/commit/cb6ef734201cfd0255c8796f1bbe90517e4b34d2): wait up to 5 seconds for the display to disconnect; if it does, wait for it to reconnect and call `dcp_ib_set_mode`; if it doesn’t, call `dcp_ib_set_mode` anyway.

This has the obvious issue of slowing down startup by at least 5 seconds on all monitors; it also seems very fragile and timing-dependent. I’m guessing there must be a better solution, since the monitors work (most of the time) in m1n1 with 11.6 iBoot, and it doesn’t seem to wait 5 seconds for the monitor to disconnect/reconnect…  (I’ve tried to measure the time iBoot takes before starting m1n1, and it seems 11.6 is only slower than 12.1 by 900ms)

tpwrules, 2022-02-19:
I have an ACER G247HL which never worked with iBoot or m1n1's display initialization. This patch gets it working properly. For the record, it says it waited 400ms to connect, 1400 ms to disconnect, then 300ms to connect again. When I chainload, it says it waited 0ms for connection, but didn't disconnect again.

I also have an HP LA2205wg which worked sometimes, but had timing issues so I would have to reset the monitor often. This patch gets it to work all the time. On that one it says it waited 700ms to connect, but did not disconnect again.

This does introduce a 5 second delay on chainloading which is somewhat annoying. But it does mean I can actually reliably see things now! Thank you for your investigation.

marcan, 2022-03-27:
Now that we have m1n1 chainloading from the ESP, I think the right solution here is to both make stage 2 always re-probe the display (that gives it a second chance to work for everyone) and then offer a m1n1 option to both configure the display mode and optionally introduce a longer timeout. That will allow people with problematic monitors to make it work, without unconditionally increasing boot times for everyone.

pat42smith, 2022-03-27:
"option to ... configure the display mode": I like that idea. With my 4k display, currently the only available resolution is 1920x2160, which is distorted enough that I wouldn't want to use it for any length of time.

marcan, 2022-03-27:
I didn't realize it did 2:1 scaled modes like that... I can fix that in the m1n1 mode selection. Pushed as 8af401ade1a8 for the future (won't help you until we make stage2 also do this and I cut a release and you enable it though).

marcan, 2022-03-27:
Oh god while looking into this I just realized why sometimes displays don't init sometimes. We're treating the HPD flag as 32 bits but it's actually just a bool, 8 bits. The rest is uninitialized, though I didn't notice it can be nonzero until now. If the top bit of the 32 happens to be set, it gets cast to signed later and interpreted as an error, and then m1n1 thinks no monitor is connected.

marcan, 2022-03-27:
Okay, for the folks with issues: please make a backup of `/boot/efi/m1n1/boot.bin` and replace it with this one: https://mrcn.st/t/boot_display-test.bin (make sure to rename it to boot.bin).

Let me know if that improves things, and then, if you want to set the mode:

```echo display=1920x1080 >> /boot/efi/m1n1/boot.bin```

You can also use 1920x1080@60 to set the rate, or even 1920x1080@59.94 for fractional rates (but the parser is really dumb, always use two decimals).

Obviously only do the `echo` once. For each time m1n1 finds the `display=` line it'll set the mode, so if you append it multiple times you'll wind up with the mode changing repeatedly.

rockinrobstar, 2022-03-27:
Just tested on my Lenovo L27m-28 (both default and with display setting) and no luck. Can confirm the file works fine otherwise on my Dell U4320Q @ 1920x1080. 

One thing I did notice is that the Lenovo does have a very aggressive sleep mode (~ 2 sec) - so not sure if that is a factor.

patlux, 2022-03-27:
I have the same problem with my Xiaomi Curved Monitor 34".

I tired all variations but it doesn't help:

```sh
cd /boot/efi/m1n1
sudo su
cp boot.bin boot.bin.bkp
rm boot.bin
wget https://mrcn.st/t/boot_display-test.bin

# first try
cp boot_display-test.bin boot.bin
reboot

# 2.
cp boot_display-test.bin boot.bin
echo display=1920x1080 >> /boot/efi/m1n1/boot.bin
reboot

# 3.
cp boot_display-test.bin boot.bin
echo display=1920x1080@60 >> /boot/efi/m1n1/boot.bin
reboot

# 4.
cp boot_display-test.bin boot.bin
echo display=1920x1080@59.94 >> /boot/efi/m1n1/boot.bin
reboot

# 5.
cp boot_display-test.bin boot.bin
echo display= auto >> /boot/efi/m1n1/boot.bin
reboot
```

VinDuv, 2022-03-27:
The modified `boot.bin` also fails to initialize the monitor in my case. I replaced it with a freshly compiled `m1n1.bin` to be able to recover m1n1 logs.

- Without `display=` option, stage2 m1n1 detects that the display is already initialized (by stage1 m1n1) and does nothing.
- With `display=auto`, stage2 m1n1  seems to correctly re-perform the DCP init:

```
Found a variable at 0x802318000: display=auto
dart: dart /arm-io/dart-dcp at 0x23130c000 is a t8020 (locked)
dart: dart /arm-io/dart-disp0 at 0x231304000 is a t8020
rtkit(dcp): booting with version 12
rtkit(dcp): unknown system endpoint 0x0a
rtkit(dcp): pre-allocated buffer (ep 0x1, dva 0xf00237000, phys 0x9e6213000)
EPIC: started interface 0 (disp0-service)
display: waited 0 ms for display status
display: connected:1 timing_cnt:45 color_cnt:12
display: timing mode: valid=1 1920x1080 60.00 Hz
display: color mode: valid=1 colorimetry=1 eotf=1 encoding=1 bpp=32
display: swapped! (swap_id=3)
Found a variable at 0x80231800d: chosen.asahi,efi-system-partition=b85d68d6-f947-4a31-8d76-cc46bca458a6
No more payloads at 0x802318054
``` 

It looks like the chainload is so fast, the monitor hasn’t started its reset cycle yet (`waited 0 ms for display status`), so it doesn’t prevent the monitor disconnection.

@tpwrules @rockinrobstar @patlux Something you can try (It seems to work in my case): Turn off the Mac, hold the power button until the boot menu shows up, and select Asahi Linux. The Mac reboots when you do this, but in my case the monitor doesn’t have the time to fully go to sleep, so the DCP initialization by m1n1 works.

patlux, 2022-03-27:
@VinDuv tried the steps you described it but it didn't helped. Neither with nor without display=auto.

VinDuv, 2022-03-27:
I started working on adapting my previous “wait for display reset” patch to enable it conditionally in m1n1: https://github.com/AsahiLinux/m1n1/pull/183
You can download a build [here](https://vincent.duvert.net/divers/boot.bin). The `display=wait` parameter is already appended so it should work directly once copied in the EFI partition.

patlux, 2022-03-27:
@VinDuv really nice work. Now it works but now it freezes at the „Running proxy“ line.

[image]

VinDuv, 2022-03-27:
@patlux Don’t reinstall! This is normal behavior. The `boot.bin` file I provided doesn’t include the U-Boot payload, so once m1n1 is started it waits for a payload to be sent over serial (from another computer). I’ll try to build a full version later; in the meantime, you can restore your original `boot.bin` from macOS.

Skirmisher, 2022-03-27:
@marcan:
> Okay, for the folks with issues: please make a backup of `/boot/efi/m1n1/boot.bin` (in the same folder) and replace it with this one: https://mrcn.st/t/boot_display-test.bin (make sure to rename it to boot.bin).

This works for me, but only if I append `display=1280x1024` (Apple Studio Display, 17" LCD, ADC). This tracks with the symptoms I was seeing (monitor LED was turning on, but no signal appeared).

marcan, 2022-03-27:
None of these displays "work" under macOS because Apple disabled iBoot display output entirely since 12.0 ;-)

There is no display output until the OS driver loads, which means no verbose boot, no iBoot messages, no Apple logo and no progress bar until the kernel is half booted, etc. In fact, you get those features back if you chainload macOS under m1n1 and the display was initialized properly.

Once Linux has a real display driver it will work like macOS. The problem we're trying to solve in this bug is bootloader display output, which is handicapped by the iBoot-mode interface that Apple provides for it. That's why we're having so much trouble with modes and timeouts and such.

alyssarosenzweig, 2022-03-27:
> Once Linux has a real display driver it will work like macOS.

*coughs* it works for me

pat42smith, 2022-03-27:
@marcan: Setting the display mode to 1920x1080 with your modified boot.bin works, thanks!

Setting 3840x2160 partially works - the top 2/3 of the screen seem fine, but the bottom 1/3 is just black.

On startup blankness - what I'm seeing is that often after a reboot (including 'shutdown -r now' in Linux), the screen will stay blank. Not always, but often. The only pattern I'm seeing is that the only times it didn't stay blank were when I had the display mode set with the new boot.bin. But even then, sometimes it stayed blank. So I don't know.

On the other hand, every time I completely shutdown the system and then restart it, the display works. Including one reinstall where I got to the "press enter to reboot" prompt inside Mac Recovery, I instead shut down the system from the apple menu. So for me at least, this is not a showstopper.

tpwrules, 2022-03-28:
marcan's patch lets me set my HP LA2205wg to its native resolution and get rid of the obnoxious nag, and for that I am grateful. But it doesn't fix the Acer. I did find on the HP that configuring it in its setup menu to always use the DVI input and disable auto-searching made it work 100% of the time, so if your monitor lets you do that it may help.

@VinDuv it would probably be good for your patch to add a configurable delay AND resolution

marcan, 2022-03-28:
4K is not currently supported since the framebuffer is not big enough to support that mode on the Mini.

VinDuv, 2022-03-29:
@patlux can you try [boot_display_wait.bin](https://vincent.duvert.net/dropbox/boot_display_wait.bin)? It should wait for the monitor to come up and then boot Linux.
@tpwrules I followed your suggestion of adding a way to control the wait delay; you can test it by downloading either `m1n1.bin` or `boot.bin` from https://vincent.duvert.net/dropbox/ and append a `display=...` line to them. The format is:

| Parameter | |
|-----------|-|
| &lt;none&gt; | Display is initialized by stage2 m1n1 and if only if it wasn’t initialized in stage1 |
| `display=auto` | Display is (re-)initialized by stage2 m1n1 regardless of its current state |
| `display=1280x1024` | Display is (re-)initialized by stage2 m1n1, preferring 1280x1024 resolution |
| `display=wait`<br> `display=wait,auto` | Display is (re-)initialized by stage2 m1n1 after waiting 10 seconds for monitor reset |
| `display=wait:6`<br> `display=wait:6,auto` | Display is (re-)initialized by stage2 m1n1 after waiting 6 seconds for monitor reset |
| `display=wait,1280x1024` | Display is (re-)initialized by stage2 m1n1 after waiting 10 seconds for monitor reset, preferring 1280x1024 resolution |
| `display=wait:6,1280x1024` | Display is (re-)initialized by stage2 m1n1 after waiting 6 seconds for monitor reset, preferring 1280x1024 resolution |

@Skirmisher if you’re interested, you can download [m1n1_1280_modes.bin](https://vincent.duvert.net/dropbox/m1n1_1280_modes.bin) that will dump all the timing modes that the DCP thinks are supported by the monitor (after setting it to 1280x1024). On my monitor it detects some resolutions that are probably not supported (3840x2160 on a 1080p monitor…), and I guess it’s the same in your case…

Skirmisher, 2022-03-29:
@VinDuv: 
> @Skirmisher if you’re interested, you can download [m1n1_1280_modes.bin](https://vincent.duvert.net/dropbox/m1n1_1280_modes.bin) that will dump all the timing modes that the DCP thinks are supported [...]

Yeah, you could say that...
```
display: waited 0 ms for display connected
display: connected:1 timing_cnt:18 color_cnt:3
display: timing mode: valid=1 1280x1024 60.02 Hz
display: color mode: valid=1 colorimetry=1 eotf=1 encoding=1 bpp=32
display: swapped! (swap_id=4)
fb init: 1280x1024 (30) [s=1280] @0xbe3a18000
fb console: max rows 60, max cols 64
fb: display logo
mode list:
 - valid=1 720x480 59.94 Hz
 - valid=1 720x480 60.00 Hz
 - valid=1 1280x720 59.94 Hz
 - valid=1 1280x720 60.00 Hz
 - valid=1 720x576 50.00 Hz
 - valid=1 1280x720 50.00 Hz
 - valid=1 1920x1080 59.94 Hz
 - valid=1 1920x1080 60.00 Hz
 - valid=1 1920x1080 50.00 Hz
 - valid=1 3840x2160 23.98 Hz
 - valid=1 3840x2160 24.00 Hz
 - valid=1 3840x2160 25.00 Hz
 - valid=1 3840x2160 29.97 Hz
 - valid=1 3840x2160 30.00 Hz
 - valid=1 3840x2160 50.00 Hz
 - valid=1 3840x2160 59.94 Hz
 - valid=1 3840x2160 60.00 Hz
 - valid=1 1280x1024 60.02 Hz
```
(I'm assuming the last line is just the manually-set mode)

patlux, 2022-03-30:
> @patlux can you try [boot_display_wait.bin](https://vincent.duvert.net/dropbox/boot_display_wait.bin)? It should wait for the monitor to come up and then boot Linux. 

It works :) Thank you!

pat42smith, 2022-04-01:
@VinDuv: your boot_display_wait.bin fixed the issue I was seeing with the screen sometimes staying blank after a reboot.

jannau, 2022-04-03:
For higher resolutions like 4k the framebuffer has to be reallocated. I have work in progress branches for that in https://github.com/jannau/m1n1/tree/display_fb_alloc and https://github.com/jannau/m1n1/tree/display_fb_alloc2
The main outstanding problem is that macOS crashes with those changes.

datajournalist12, 2022-04-04:
@VinDuv what folder should I direct boot_display_wait.bin into? Sorry if that's a dumb question; I'm very new to Linux.

GitHuhl, 2022-04-13:
Hello everyone. First of all thanks so much marcan and Alyssa and everyone else for getting this kickass distro off the ground so quickly! 

My monitor is a Samsung, most easily found with this name: LU32J590UQNXZA. I opened an issue about the blank screen, didn't see/know this one may be relevant. I haven't tried any of the suggested fixes yet.

I can get Asahi to output graphics by setting the monitor into picture-in-picture/picture-by-picture mode (annoyingly needing to use another machine to activate the monitor before turning on my Mini) -- but otherwise, flawless experience so far despite only having half my screen. It works just well enough to where I'm tempted to just wait for the fixes to be stable and get them via pacman when they come out.

marcan, 2022-04-13:
@GitHuhl That sounds like the same issue then; your monitor likely goes through a hotplug cycle when it wakes up. Using PiP mode keeps it awake and stops it from doing that.

Looks like we'll have to add an option to the installer to pre-configure m1n1 with a higher display time-out, for folks with this issue. I wonder if we can make this automagic; maybe m1n1 can report whether such a hotplug happened / the safe display init time in the device tree somehow. Then the installer can initially configure it with a large time-out, and the firstboot script can drop it down to something faster if it detects a safe monitor...

VinDuv, 2022-04-17:
I like the idea of auto-configuration, but I think it will be tricky to implement.  My monitors temporarily disconnect upon waking up only if they have been asleep for a few seconds. (I guess they need to disconnect only to get out of “deep sleep”, and they don’t “deep sleep” immediately upon losing signal from the computer?)
So that means that when m1n1 starts for the first time after rebooting from macOS, the monitor will not disconnect, m1n1 won’t detect hotplug and the first boot script will disable the wait…

So I think the only way to have an auto-detection working would be to have a special “display check” mode in m1n1 that would:
1. Turn off the display, since stage1 m1n1 will have turned it on (assuming the DCP iBoot interface allows doing this)
2. Wait some amount of time for the display to go to sleep (at least 10 seconds for the E2473HDS)
3. Turn the display back on. Wait to see if it disconnects.

That would probably work, but having the display go to sleep in m1n1 for 10 seconds (probably more) would be worrying for the user, I think.

Maybe a better alternative would be to do the detection in the installer, from macOS? This would also require temporarily turning off the display, but at least the installer could warn about it beforehand…

I’ve done some preliminary testing and it looks like it’s possible to turn the display off (with `pmset displaysleepnow`), back on (with `caffeinate -u`) and get display properties (with the `AppleCLCD2` IOKit class). I’ll see if it’s possible to do this from Python…

VinDuv, 2022-05-07:
Update: I wrote a Python script that detects, from macOS, if a monitor disconnects when getting out of sleep mode. It also detects the monitor native resolution.

I’d like to confirm that it works with other people’s hardware before attempting to integrate it into the installer, so if you can, please run it and give me the results :)

The script is [here](https://vincent.duvert.net/dropbox/display-info.py) and requires Python 3 to be installed in macOS. If you haven’t Python installed, you can use the following Terminal command that will download Python (from the Asahi Installer) and run the script directly:
```
curl https://vincent.duvert.net/dropbox/display-info.sh | sh
```

The script doesn’t require any special privileges; you can run it from a guest account. You can also run it directly from the recovery environment.

Thanks!

DarkShadow44, 2022-05-07:
I'm one of the guys who needs "boot_display_wait.bin". My results are:

```
Mac model: Macmini9,1
disp0,t8103: external (U3277WB), connected, HDMI transport, 3840x2160, did not disconnect
dispext0,t8103: external, disconnected
```

pat42smith, 2022-05-08:
@VinDuv: My screen was often staying blank after a reboot (but was ok after
shutdown & start up). Running your display-info.py produces:

Mac model: Macmini9,1

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

dispext0,t8103: external, disconnected

There is only one attached display, connected by HDMI as reported in the
disp0 line. The resolution is correct.

VinDuv, 2022-05-08:
Okay, so it looks like the test didn’t detect the disconnection :-/ Weird.
Some things I though about:
 - Maybe the display actually disconnects after the test completes. The test duration was 10 seconds, I changed it to 20.
 - My displays only disconnect on wakeup if they were asleep for more than ~10 seconds. The test currently waits 20 seconds, but maybe some displays need more? (In @pat42smith ’s case, it looks like it’s the reverse: the display only disconnects if it has been sleeping for a small duration, which only happens during reboots). To check this, I made the sleep delay configurable:
   * If the script is started without arguments, it will prompt for the duration (with a default of 20 seconds).
   * The script can be started with one or more duration arguments, and will run the test for all the durations specified. For instance, `display-info.py 5 10 30` will perform the test after sleeping the display for 5 seconds, then perform it again after sleeping for 10 seconds, etc. To do this from `curl`, add the `-s` option to `sh`:
    `curl https://vincent.duvert.net/dropbox/display-info.sh | sh -s 5 10 30`.

DarkShadow44, 2022-05-08:
```
Test results:
Mac model: Macmini9,1
Sleep duration: 5.0
disp0,t8103: external (U3277WB), connected, HDMI transport, 3840x2160, did not disconnect
dispext0,t8103: external, disconnected
Detection in progress, please wait…
Test results:
Mac model: Macmini9,1
Sleep duration: 10.0
disp0,t8103: external (U3277WB), connected, HDMI transport, 3840x2160, disconnected after 8479ms for 667ms
dispext0,t8103: external, disconnected
Detection in progress, please wait…
Test results:
Mac model: Macmini9,1
Sleep duration: 30.0
disp0,t8103: external (U3277WB), connected, HDMI transport, 3840x2160, disconnected after 1975ms for 2656ms
dispext0,t8103: external, disconnected
```

pat42smith, 2022-05-08:
From display-info.py 5 10 30:

Mac model: Macmini9,1

Sleep duration: 5.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

dispext0,t8103: external, disconnected

Detection in progress, please wait…

Test results:

Mac model: Macmini9,1

Sleep duration: 10.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 3875ms for 409ms

dispext0,t8103: external, disconnected

Detection in progress, please wait…

Test results:

Mac model: Macmini9,1

Sleep duration: 30.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

dispext0,t8103: external, disconnected

(leaves me wondering if the disconnection happens by pure random chance?)
I tried again at multiples of 2 seconds. The results, showing just the
changing lines:

Sleep duration: 2.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 4.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9848ms for 426ms

Sleep duration: 6.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 7872ms for 368ms

Sleep duration: 8.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 5822ms for 411ms

Sleep duration: 10.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 12.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 1799ms for 470ms

Sleep duration: 14.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 16.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 18.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 20.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 22.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 24.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 26.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 28.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

Sleep duration: 30.0

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

I noticed one more thing, that may not matter. Most of the time the test
was running, I just left my system alone and didn't really watch as the
screen repeatedly blanked and came back. But once (I think during the 24
sec test) I noticed the terminal window was displaying a message along the
lines of "Displays should be blank now. Are they?". I moved the mouse, but
the on screen cursor did not move. Then I typed "y" and the message went
away.

vtsingaras, 2022-05-10:
Mac model: Macmini9,1
Sleep duration: 20
disp0,t8103: external (Mi Monitor), connected, HDMI transport, 3440x1440, disconnected after 9678ms for 1425ms
dispext0,t8103: external, disconnected

VinDuv, 2022-05-14:
@DarkShadow44: okay, looks like the script successfully detected the disconnection at 10 seconds, which is a bit weird because it should have worked the first time you tried it… Maybe there are some timing issues. Could you run the test multiple times with a 11 second delay and check if it says “disconnected after XXXX” every time?

@pat42smith The “Displays should be sleeping now” message is printed just before display sleep, and erased just after display wake-up. If you can see it on your monitor it means the monitor isn’t sleeping when it should, and that means the
test results will be wrong. I just changed the message to be more clear about that. Unexpected displays wake-ups may happen if an application decides to force the displays on during the test (or if you move the mouse/press a key) and I haven’t yet found a good way to detect or prevent that so displaying a message is all the test can do for now.

Regarding your test results, they look a bit inconsistent but they seem to confirm that your display only disconnects after short sleeps. Maybe try to run the test multiple times with a small delay (3 seconds) and see if it consistently says “disconnected after XXXX”?

@vtsingaras Thanks for your test result! It looks like your monitor starts disconnecting very late, close to the 10 seconds mark, so I’ll probably extend the wait delay a bit in the m1n1 patch.

DarkShadow44, 2022-05-14:
Seems to disconnect consistently now... Tried 10 times with 11s.

pat42smith, 2022-05-14:
@VinDuv Regarding the "should be sleeping now" message: I was deliberately
not using the keyboard or mouse during the test. But as I recall, I was
doing a bit of writing on the same table - so the table might have moved a
bit, causing the mouse to register a movement. I don't know, but it seems
plausible.

Result of 10 tests at 3 seconds:

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10835ms for 449ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10919ms for 351ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10933ms for 351ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10840ms for 429ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10931ms for 331ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10826ms for 453ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10911ms for 370ms

10 tests at 4 seconds:

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9823ms for 459ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9828ms for 453ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9842ms for 431ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9853ms for 412ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9813ms for 461ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9886ms for 392ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 9944ms for 340ms

The display has a feature to blank the screen when no one is sitting in
front of it. I disabled that feature and tried again 10 times at 3 seconds:

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10906ms for 330ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10825ms for 465ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10900ms for 371ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10864ms for 398ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10846ms for 414ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10903ms for 363ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 10890ms for 398ms

Finally, 10 tests at 2 seconds:

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 11939ms for 340ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 11845ms for 438ms

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, did not disconnect

disp0,t8103: external (VP2785 SERIES), connected, HDMI transport,
3840x2160, disconnected after 11916ms for 362ms

I really don't know what to make of all this. Maybe the install script
should offer the option to shutdown instead of reboot?

DarkShadow44, 2022-05-15:
How can it disconnect after > 10s when your test time 4, 3, and 2?

tpwrules, 2022-05-15:
Here are the results for my two monitors:
```
% curl https://vincent.duvert.net/dropbox/display-info.sh | sh 
[...]
Monitor sleep duration: [20] 30
Detection in progress, please wait…
Test results:
Mac model: Macmini9,1
Sleep duration: 30.0
disp0,t8103: external (LA2205), connected, DVI transport, 1680x1050, disconnected after 1378ms for 614ms
dispext0,t8103: external, disconnected

% curl https://vincent.duvert.net/dropbox/display-info.sh | sh
[...]
Monitor sleep duration: [20] 
Detection in progress, please wait…
Test results:
Mac model: Macmini9,1
Sleep duration: 20
disp0,t8103: external (G247HL), connected, HDMI transport, 1920x1080, disconnected after 2479ms for 404ms
dispext0,t8103: external, disconnected```

It's worth nothing that the first monitor (the LA2205) I can get to stop disconnecting by turning off auto source switching in its menu. It is also a 1680x1050 native panel, but m1n1 currently detects it as 1920x1080. It can render that resolution fine, but it displays an unremovable dialog box whining of a non-optimal resolution. Please be sure to have a way for the user to select the monitor's resolution in the installer.

VinDuv, 2022-06-04:
A better solution to the disconnect problem was recently found: It basically disables the display controller once the display settings have been configured, so if the monitor disconnects and reconnects, it will still receive data.
This removes the need to wait for disconnection and speeds up booting. It also makes input switching work on all monitors.
This has been integrated into m1n1 (https://github.com/AsahiLinux/m1n1/commit/714420a694170a293767e6e9a1daccd6274ac938) but has not yet been released; if you want to test it, you can download this [boot.bin](https://drop.duvert.net/boot.bin) and put it in your EFI partition.
Note that the current solution may still have issues if the monitor disconnects at a very specific time; a fix for this is in the works.
If your monitor’s resolution is less than 1920x1080, you will still need to add a `display=<res>` line at the end of the `boot.bin` file. I’m currently working on a patch for the installer to do this automatically (and allow changing a previous setting).

pat42smith, 2022-06-05:
> This has been integrated into m1n1 (714420a

@VinDuv: I installed your new boot.bin and rebooted a few times. It appears
this fixed my "blank screen after reboot" problem. Thanks!

janrinze, 2022-06-10:
Thanks for all these pointers. Finally managed to get 4K output to my monitor!

xfim, 2022-08-02:
Dear all,

I tried the `boot.bin` linked by @VinDuv on June 4th, but the machine does not work anymore with the kernel that I was using. It freezes during boot. Noticeably, during boot it still says that is expecting 1980x1080, although my monitor is 4K. Do I need anything newer? With such `boot.bin` do I still have to append `display=auto` to it? I am on gentoo combining asahi-sources.

Thank you,

janrinze, 2022-08-03:
@xfim: this thread is old, that boot.bin does not match newer setups and the related changes have already been merged in the latest m1n1.

Did you backup the original boot.bin? use that one instead and also keep an original copy. 

Restore the boot.bin from your backup and append ```display=3840x2160```  for a 4k monitor. (assuming that's the resolution of your display)

For anyone else here trying to use old boot.bin files linked here, please don't!
Just install Asahi Linux as described by the project.

xfim, 2022-08-04:
OK, understood @janrinze , thank you for the clarification.
Fortunately, I made a backup and I can go back. The problem is that appending the resolution did not work, and that's why I tried it. But I will make it work.

chadmed, 2025-05-31:
This should no longer be relevant

closed 2025-05-31 by chadmed