AsahiLinux/m1n1 #176: [Question] Can I dual boot Linux and Mac OS with asahilinux? Like switching between them when I want

Galorlx, 2022-03-22:


rowanG077, 2022-03-22:
Yes you can choose between Linux and OSX in the bootpicker. See to the [FAQ](https://asahilinux.org/2022/03/asahi-linux-alpha-release/#can-i-dual-boot-macos-and-linux).

Galorlx, 2022-03-22:
When would you guess Linux could support like full powerful usability

Galorlx, 2022-03-22:
> Yes you can choose between Linux and OSX in the bootpicker. See to the [FAQ](https://asahilinux.org/2022/03/asahi-linux-alpha-release/#can-i-dual-boot-macos-and-linux).

When would you guess Linux could support like full powerful usability

rowanG077, 2022-03-22:
I'm not the person to ask. But there is no official estimate, it's done when it's done. There are people working on userspace drivers.