AsahiLinux/m1n1 #177: [HELP] error with creating linux

Galorlx, 2022-03-22:
[image]

WP-LKL, 2022-03-22:
What is the partition size you are allocating and how much is avaliable?

Hello9999901, 2022-03-23:
I had this error. the problem is that diskutil is a little buggy with this (idk why). you just need to allocate more storage to the Asahi linux storage. I am sure there is a better fix but that is the way I fixed it. I have a 1 TB drive and partitioned 50% to it. Asahi said 38 percent was good for whatever reason, but diskutil clearly wasn't taking it. I didn't want to troubleshoot for hours and I had enough storage to spare so that could be potential solution :) 
sorry I couldn't be of much help

WP-LKL, 2022-03-23:
> I had this error. the problem is that diskutil is a little buggy with this (idk why). you just need to allocate more storage to the Asahi linux storage. I am sure there is a better fix but that is [...]

Just to add. As described on Twitter some of these installs have been revealing latent corruptions on the disks in some scenarios. I will edit my comment if I find the fix mentioned in the same thread.

Galorlx, 2022-03-23:
> Just to add. As described on Twitter some of these installs have been revealing latent corruptions on the disks in some scenarios. I will edit my comment if I find the fix mentioned in the same [...]

My partition is on my main Mac it had to 177GB and I did min (didn’t work) I tried max as the options and they didn’t work I’ll try to do more later. I screwed up the installation before and I did the same numbers and it worked but now it doesn’t. Like I said I’ll try again later

Galorlx, 2022-03-23:
> Just to add. As described on Twitter some of these installs have been revealing latent corruptions on the disks in some scenarios. I will edit my comment if I find the fix mentioned in the same [...]

You have anything on that thread?

marcan, 2022-03-23:
Please attach the log file at `/tmp/asahi-installer/installer.log` so we can take a look.