AsahiLinux/m1n1 #184: is it possible to install Ubuntu with "UEFI Environment only" option?

ocowala, 2022-03-30:
Hey,
I was thinking on testing this new technology out. I want to install ubuntu instead of arch linux and i think using the uefi environment option would make it work. however, i need to know how to boot into it and check if it is possible with m1 macbook pro architecture. Could you guys help me out here?

lennylxx, 2022-04-03:
Someone already posted a Asahi Installer w/ Debian installer, and I successfully change my Debian installation to Kali afterwards. This is possible because Kali is just based on Debian. You could try to do similar thing to convert Debian to Ubuntu, but since these two systems are more different than Debian v.s. Kali, you might encounter unrecoverable failure.

00ihsan, 2022-09-01:
How did you install Kali?

00ihsan, 2023-02-01:
> Someone already posted a Asahi Installer w/ Debian installer, and I successfully change my Debian installation to Kali afterwards. This is possible because Kali is just based on Debian. You could [...]

Do you have a tutorial for kali?