AsahiLinux/m1n1 #196: Serial crash when trying to boot t6000 dev kernel with hypervisor

rockinrobstar, 2022-04-17:
I am currently trying to boot my MBP 14 Pro (target) from my M1 Mini (proxy) using the KDK Version 12.3 (21E230) & the following command sequence:
```
export M1N1DEVICE=/dev/cu.usbmodemP_01
./proxyclient/tools/chainload.py -r build/m1n1.bin
python3 proxyclient/tools/run_guest.py -S -m hv/trace_all.py -s ../macOS.kernel/t6000.21E230.DWARF ../macOS.kernel/dev.kc.macho.development -- "cpus=1 debug=0x14e serial=3 apcie=0xffffffef -enable-kprintf-spam wdt=-1"
```

The kernel starts to boot but after a few seconds it crashes with the following output (and the MBP rebooting):
```
[cpu0][0xfffffe00138efd28] MMIO: W.4   0x291e6f0ec (unknown, offset 0x291e6f0ec) = 0x0
[cpu0][0xfffffe001460e688] MMIO: R.4   0x2920a1300 (nub-spmi0[0], offset 0x00) = 0x1000100
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<console>", line 1, in <module>
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/shell.py", line 129, in display
    val()
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/hv.py", line 1440, in start
    self.p.hv_start(self.entry, self.guest_base + self.bootargs_off)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 975, in hv_start
    return self.request(self.P_HV_START, entry, *args)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 643, in request
    return self._request(opcode, *args2, **kwargs)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 609, in _request
    reply = self.iface.proxyreq(req, reboot=reboot, no_reply=no_reply, pre_reply=None)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 356, in proxyreq
    return self.reply(self.REQ_PROXY)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 230, in reply
    reply += self.readfull(1)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 167, in readfull
    block = self.dev.read(size - len(d))
  File "/opt/homebrew/lib/python3.9/site-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured
```
Things I have tried: 

- Increasing CPU to cpu=2+ but I get this instead:
```
[cpu1][0xfffffe0013e3ecec] MMIO: R.1   0x59020000e (apcie[0], offset 0x20000e) = 0x80
Event checksum error: Expected 0x00150100, got 0xd0decade
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.9/3.9.12/Frameworks/Python.framework/Versions/3.9/lib/python3.9/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<console>", line 1, in <module>
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/shell.py", line 129, in display
    val()
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/hv.py", line 1440, in start
    self.p.hv_start(self.entry, self.guest_base + self.bootargs_off)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 978, in hv_start
    return self.request(self.P_HV_START, entry, *args)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 646, in request
    return self._request(opcode, *args2, **kwargs)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 612, in _request
    reply = self.iface.proxyreq(req, reboot=reboot, no_reply=no_reply, pre_reply=None)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 359, in proxyreq
    return self.reply(self.REQ_PROXY)
  File "/Users/robin/Developer/m1n1/proxyclient/m1n1/proxy.py", line 259, in reply
    raise UartChecksumError()
m1n1.proxy.UartChecksumError
```
- Increasing iface.dev.timeout = 1 in proxy.py (no change).
- Using trace_pmgr.py instead of trace_all.py (no change). 
- Setting the baud to 256000 using `export M1N1DEVICE=/dev/cu.usbmodemP_01:256000` (no change)

marcan, 2022-04-17:
That means the guest is panicking and rebooting the machine. The hypervisor makes no attempt to stop the guest from doing that; it is a thin layer, not a security boundary. So if the guest kills the machine, the hypervisor goes along with it.

You should look at the serial log from macOS (available on the secondary USB ACM interface) to figure out why it's doing that. 

Are you perhaps trying to boot macOS from a m1n1 proxy installed via the Asahi Linux installer? That won't work - to boot macOS you need to be running the OS volume macOS was installed under, which means you need to install m1n1 with kmutil/bputil on top of an existing macOS install. If you try to boot the macOS kernel under an Asahi Linux volume it'll just panic when it can't mount the root filesystem.

rockinrobstar, 2022-04-17:
Yes, you caught me red handed, and now that you have explained it - it makes more sense. I will ask some more questions on irc about installing a 2nd macOS Volume (avoiding touching my original macOS install for now) and will attempt to update the wiki so others don't follow my folly. Thanks for your quick reply.

closed 2022-04-17 by rockinrobstar