AsahiLinux/m1n1 #215: proxyclient ADT decoder faults on pmgr devices with iPhone 5S/6 ADT

konradybcio, 2022-06-02:
Example output:

```
python3 -m m1n1.adt ~/DeviceTree.n61ap.bin > n61_m1n1parsed
Exception parsing /device-tree/arm-io/pmgr.devices value 8000020100000000000000000500000035de000070fd34000000000000000000435055300000000000000000000000008000020200000000000001000600000035de000070fd3400000000000000000043505531000000000000000000000000100002030000000000000000000000000000000000000000000000000000000043505500000000000000000000000000030000040000000000001f01000000000000000000000000000000000000000053494f5f500000000000000000000000000000060000000000000902000000000000000000000000010000000c00000055534200000000000000000000000000000000080600000000000a0201000000991903000c62000000000000000000005553424354524c0000000000000000000000000a0800000000000b02000000000000000000000000000000000000000055534232484f535430000000000000000000000c0800000000000d02000000000000000000000000000000000000000055534232484f535431000000000000000000000e0800000000000f02000000000000000000000000000000000000000055534232484f5354320000000000000020000010000000000b00150201000000b8be0600000000000100000007000000444953505f4255534d55580000000000200000120000000005001b020100000031280000146e010001000000010000004d4544494100000000000000000000002300001600000000190003030100000066260a003dea01000100000000000000414e5300000000000000000000000000000000180000000000001a02040000000000fa000a971300010000000600000049535000000000000000000000000000200000191200000007001c020100000099d96200c420040001000000020000004d5352000000000000000000000000002000001a1200000006001d020100000028dc1500c420040001000000030000004a5047000000000000000000000000002000001b1200000009000003020000000000c8003dca0e00010000000400000056454e430000000000000000000000002000001c100000000a0016020100000066e652007ad405000100000008000000444953503000000000000000000000002300001d100000000c00190201000000f5286100494c040001000000090000004449535031000000000000000000000010000020180000000000000000000000000000000000000000000000000000004953505f44415254000000000000000010000021190000000000000000000000000000000000000000000000000000004d53525f444152540000000000000000100000221a0000000000000000000000000000000000000000000000000000004a50475f444152540000000000000000100000231b0000000000000000000000000000000000000000000000000000004156455f444152540000000000000000100000241c00000000000000000000000000000000000000000000000000000044495350305f44415254000000000000100000251d00000000000000000000000000000000000000000000000000000044495350315f4441525400000000000010010126000000000000000004000000000050000a97130000000000000000004953502d46415354000000000000000010010127000000000000000002000000000064003dca0e00000000000000000056454e432d464153540000000000000010010128000000000000000003000000a330110028dc02000000000000000000564445432d46415354000000000000001001012900000000000000000100000099192200c420040000000000000000004a50472d464153540000000000000000000002300000000000000800000000000000000000000000000000000000000043504d0000000000000000000000000030000031000000000100000000000000000000000000000000000000000000004c494f5f4146000000000000000000002300003200000000000000010100000078e90000ba69020000000000000000004c494f000000000000000000000000000300003300000000000001010000000000000000000000000000000000000000494f4d5558000000000000000000000003000034000000000000020100000000000000000000000000000000000000004149430000000000000000000000000003000035000000000000030100000000000000000000000000000000000000004445425547000000000000000000000003000036000000000000040100000000000000000000000000000000000000004457490000000000000000000000000003000037000000000000050107000000991919000000000000000000000000004750494f00000000000000000000000024000038000000001a0004020000000000000000000000000000000000000000504349455f5245460000000000000000200000390000000004000103010000001ec51300353e0100010000000d000000504349450000000000000000000000000000003a00000000000002030000000000000000000000000000000000000000504349455f415558000000000000000000000040040000000000060101000000000006000000000000000000000000004d43413000000000000000000000000000000041040000000000070101000000000006000000000000000000000000004d43413100000000000000000000000000000042040000000000080101000000000006000000000000000000000000004d43413200000000000000000000000000000043040000000000090101000000000006000000000000000000000000004d434133000000000000000000000000000000440400000000000a0101000000000006000000000000000000000000004d434134000000000000000000000000000000450400000000000b01000000000000000000000000000000000000000050574d30000000000000000000000000000000460400000000000c01000000000000000000000000000000000000000049324330000000000000000000000000000000470400000000000d01000000000000000000000000000000000000000049324331000000000000000000000000000000480400000000000e01000000000000000000000000000000000000000049324332000000000000000000000000000000490400000000000f010000000000000000000000000000000000000000493243330000000000000000000000000000004a04000000000010010000000000000000000000000000000000000000535049300000000000000000000000000000004b04000000000011010000000000000000000000000000000000000000535049310000000000000000000000000000004c04000000000012010000000000000000000000000000000000000000535049320000000000000000000000000000004d0400000000001301000000000000000000000000000000000000000053504933000000000000000000000000030000500400000000001401000000000000000000000000000000000000000055415254300000000000000000000000030000510400000000001501000000000000000000000000000000000000000055415254310000000000000000000000030000520400000000001601000000000000000000000000000000000000000055415254320000000000000000000000030000530400000000001701000000000000000000000000000000000000000055415254330000000000000000000000030000540400000000001801000000000000000000000000000000000000000055415254340000000000000000000000030000550400000000001901000000000000000000000000000000000000000055415254350000000000000000000000030000560400000000001a01000000000000000000000000000000000000000055415254360000000000000000000000030000570400000000001b01000000000000000000000000000000000000000055415254370000000000000000000000030000580400000000001c01000000000000000000000000000000000000000055415254380000000000000000000000030000590400000000001d010000000000000000000000000000000000000000414553300000000000000000000000002300005a0400000003001e0101000000c215010000000000000000000000000053494f000000000000000000000000002000005b0c00000017000002000000000000000000000000000000000000000048534943305f504859000000000000002000005c0e00000018000102000000000000000000000000000000000000000048534943315f50485900000000000000000000600000000000000202000000000000000000000000000000000000000049535053454e53300000000000000000000000610000000000000302000000000000000000000000000000000000000049535053454e533100000000000000002300006200000000100006020100000066665500d9ce010000000000000000004d43430000000000000000000000000023000063000000000f0007020100000099190000852b010000000000000000004d4355000000000000000000000000000300006400000000000008020000000000000000000000000000000000000000414d5000000000000000000000000000000000670a00000000000c02000000000000000000000000000000000000000055534232484f5354305f4f484349000000000069080000000000110200000000000000000000000000000000000000005553424f5447000000000000000000002300006a000000000e001202010000000080210068b100000000000000000000534d58000000000000000000000000002300006b000000000d0013020100000099991500eb5101000000000000000000534600000000000000000000000000000300006c00000000000014020000000000000000000000000000000000000000435000000000000000000000000000001000006d000000000000000000000000000000000000000000000000000000004350552d425553590000000000000000130000701c00000000000000000000000000000000000000000000000000000044495350302d5600000000000000000003000071100000000000170200000000000000000000000000000000000000004d4950495f4453490000000000000000010000721c000000000018020100000000005000000000000000000000000000445000000000000000000000000000001000007400000000000000000000000000000000000000000000000000000000444953502d46422d41435449564500001000007600000000000000000000000000000000000000000000000000000000444953502d4d49452d44504200000000200000771200000008001e0203000000cccc0a0028dc02000100000005000000564445430000000000000000000000000000007c0000000000000403000000000000000000000000020000000a0b0000474658000000000000000000000000002b00007d00000000020000040000000000000000000000000000000000000000534550000000000000000000000000001000007e00000000000000000000000000000000000000000000000000000000535743505400000000000000000000003100008000000000110000000000000000000000000000000000000000000000414d504341485f4649584544000000003100008100 [... cut ...]
  File "/opt/homebrew/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/homebrew/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Volumes/Projects/m1n1/proxyclient/m1n1/adt.py", line 581, in <module>
    assert new_data == adt_data[:len(new_data)]
AssertionError
```

Similar failures pop up when using 5S (n51/53) ADTs from the latest IPSWs.

konradybcio, 2022-06-02:
These older devices have a slightly different data structure in pmgr.devices than M1, investigating..

konradybcio, 2022-06-02:
[ip6_pmgr_devices.txt](https://github.com/AsahiLinux/m1n1/files/8823426/ip6_pmgr_devices.txt)

I tried to apply the M1 data structure to this, it *somewhat* aligns...

jannau, 2022-06-03:
This seems to be difference between ios and macos and not just related to old devices.Parsing of `PMGRPowerDomains` fails for the iphone 13 ADT from the IPSW too. I didn't look further into it since I was checking the ADT for something unrelated.

marcan, 2022-06-03:
This isn't strictly an iOS thing. The ADT for iPhone13,4 (d54pap) from iOS 14.7.1 does parse correctly.

You can use the compat_match feature to specify a different parsing for your device. For example:

```
DEV_PROPERTIES = {
    "pmgr": {
        "pmgr1,foobar": {
            "devices": None,
        },
        "*": {
# ...
```

That will avoid parsing `devices` for pmgr compat `pmgr1,foobar`. Or you can declare a separate struct and specify it instead.

asdfugil, 2025-09-17:
this issue should be now resolved

closed 2025-09-18 by konradybcio