AsahiLinux/m1n1 #236: [Testing] virtual targets

sh7dm, 2022-08-04:
How hard would it be to app virtual (aarch64 QEMU) hardware support to m1n1?  This could be useful to develop and test hardware-independent features like config, filesystem and secure boot without needing the hardware, also allowing debugging. If it's simple enough, that would be useful for Rust porting as well.

marcan, 2022-08-23:
I already replied elsewhere, but for the record: this is difficult, because m1n1 is deeply tied to Apple-specific CPU and hardware features by design (its job is to deal with all that). Even if you stripped out all the CPU stuff, there's still Apple's bespoke NVMe controller to deal with, and qemu doesn't emulate any of that.

That said, debugging is already easy: just run m1n1 inside the m1n1 hypervisor. You even get a virtual UART and a gdb stub to play around with :)

sh7dm, 2022-08-23:
Thank you for elaboration. I'm closing this now, secure boot discussion should be moved over to #195 . We should consider a first generation of secure boot without SEP support, even though it's vulnerable in case somebody examines the NAND storage for leftover key encrypted with an old passphrase.

closed 2022-08-23 by sh7dm