AsahiLinux/m1n1 #241: Second Display is not detected.

yepezcode, 2022-08-23:
Hi, I have a Mac Mini and I use the HDMI port in order to my main monitor but my second display is not detected, I using an adapter type c with 2 connections HDMI and VGA in this case I use another HDMI port.

I tried reboot the system and disconnect and connect the port and still not working. 

[image]

Another thing is my monitor is 120Hz. How I can help to set 120 Hz.

jannau, 2022-08-23:
Display output via USB-C is not yet supported, see https://github.com/AsahiLinux/docs/wiki/Feature-Support. 120 Hz refresh rate depends also on the driver for the for display processor (DCP). At least on the Macbook Pro 14"/16"'s internal display 120 Hz is handled as special case, not sure if it will work on external displays differently / out of the box.

marcan, 2022-08-23:
Please see [Feature Support](https://github.com/AsahiLinux/docs/wiki/Feature-Support) for currently supported features. Secondary screens are not yet supported.

closed 2022-08-23 by marcan