AsahiLinux/m1n1 #242: Duplicate entry for register SPRR_KMASK0_EL12 in apple_regs.json

sagittarius-a, 2022-08-23:
## Context

I'm using `tools/apple_regs.json` as input for other tools. This does not create issues in `m1n1` (or at least I did not idenfity it myself).

## Issue

The `SPRR_KMASK0_EL12` register has 2 definitions such as:

line 144:
```
{"index": 0, "name": "SPRR_KMASK0_EL12",            "fullname": "SPRR Kernel Permission Unlock Mask (EL12)",                    "enc": [3, 4, 15, 6, 0  ], "width": 32},
```

line 216:
```
{"index": 0, "name": "SPRR_KMASK0_EL12",            "fullname": "SPRR Kernel Permission Unlock Mask 0 (EL12)",                  "enc": [3, 6, 15, 6, 0  ], "width": 32},
```

Since the first is defined adjacent to `SPRR_UMASK*`, I couldn't tell if it's a typo or if 2 registers are effectively used for the same thing.

I apologize in advance if this issue is not relevant :)

DragonStar373, 2026-10-05:
Petition to close? This issue has been stale for over four years now, and SPRR_KMASK0_EL12 isn't even defined in apple_regs.json anymore

sagittarius-a, 2026-10-05:
Closing since this is not relevant anymore.

closed 2026-10-05 by sagittarius-a