AsahiLinux/m1n1 #249: Enable the notch

7596ff, 2022-09-05:
According to the two blog posts that mention the notch, support for drawing on the entire screen will be added in the future. This is the commit that hides it: https://github.com/AsahiLinux/m1n1/commit/287a963bcca63b91972f5300561ef45c50959579

I have interest in this feature, and I noticed there was no issue for it yet.

marcan, 2022-09-06:
The notch will never be enabled in m1n1. m1n1 provides a boot-time framebuffer and there is no reasonable construct for specifying the notch location such that downstream software can do something with it.

Notch support has to be part of the Linux DCP driver, which is not merged yet.

closed 2022-09-06 by marcan

7596ff, 2022-09-07:
Ah, thanks for the feedback. I only made this issue here because I noticed precedent in this repository.