AsahiLinux/m1n1 #253: Q: Installing Linux on iPad 2?

Kreyren, 2022-09-19:
Referencing https://git.dotya.ml/kreyren/kreyren/issues/30

Can i use this distribution on this iPad 2? or could you provide any useful information to help me solve the issue?

Thanks!

jannau, 2022-09-19:
No, the iPad 2 uses a 32-bit Arm processor (Cortex A9) and has very little in common with Apple silicon processors starting with with M1 and A14 (only available on devices with locked bootloader and thus not in scope of this project).

marcan, 2022-09-20:
Apple's boot system does not allow installing third-party OSes on iOS devices, and we are not working on ports to any iOS device at this time.

closed 2022-09-20 by marcan

liangqi, 2023-10-06:
https://ipadlinux.org

liangqi, 2023-10-06:
https://en.wikipedia.org/wiki/Apple_A7 is the first 64bit, ARMv8-A

Kreyren, 2023-10-06:
> Apple's boot system does not allow installing third-party OSes on iOS devices, and we are not working on ports to any iOS device at this time. -- @marcan (https://github.com/AsahiLinux/m1n1/issues/253#issuecomment-1251859412)

As far as i was told there is a checkm8 exploit that allows full write access and unlocking of the bootloader https://github.com/axi0mX/ipwndfu

So the issue should be only linux compatibility?