AsahiLinux/m1n1 #254: "Second-generation" Apple silicon OpenBSD support

amada95, 2022-09-22:
Hello! Quick disclaimer, this is probably a pretty low priority "enhancement" question, so feel free to tag it as you see fit. 

I greatly appreciate the work you all are doing with apple silicon platforms, and I've seen that you've been collaborating with the OpenBSD developers - they seem to now (with some confidence) support Apple M1/M1 Pro/M1 Max chips. I'm not asking for an ETA for "full support", as I'm well aware that's a silly question, but as you continue to work on Linux support for the "Second-generation" platforms (i.e. M2, M1 Ultra), will support also be mirrored with OpenBSD?

Thank you, and have a great day!

jannau, 2022-09-23:
Please do not use m1n1 issues to inquire about the status of other software.

The M1 Ultra is not a "second gen" chip. It might be already supported. Linux needed zero code changes for M1 Ultra support. On M2 support is being worked on as far as I'm aware.

amada95, 2022-09-23:
I'm not inquiring about the status of other software - I'm specifically asking on the m1n1 repository *because* it is a necessary part of booting "untrusted" operating systems due to the Apple silicon platforms' lack of EFI. My question pertained to continued collaboration with the OpenBSD team concerning this bootloader and shared knowledge of the platform between Asahi and OpenBSD, not OpenBSD feature support.

marcan, 2022-09-24:
The OpenBSD team are aware of the M2 and what is needed to support it, and we continue to work on things time and hardware permitting I assume. But this really doesn't have anything to do with this repository. m1n1 doesn't care what OS you run under it.

closed 2022-09-24 by marcan

amada95, 2022-09-30:
I see - I read the wiki pertaining to the boot process for AS macs, and I think I have a bit of a better understanding of m1n1 now. Thanks for the continued excellent work!

p.s. a little tweaking got OpenBSD to load from m1n1, not bad for a black box :)