AsahiLinux/m1n1 #258: Cannot build with CHAINLOADING=1

SuperKenVery, 2022-10-23:
I installed the aarch64-unknown-none-softfloat tool chain and set it as default. When compiling I get errors like

``` shell
can’t find crate for `core `
```

and many others. 

My command:
`make CHAINLOADING=1`

If I add ARCH=arm64, the c compiler fails (arm64gcc: no such file or directory)

If I cd into the vendor directories and run cargo build, they build individually. But when I run make, it fails. 

My setup:
Asahi Linux, M1
Kernel 5.19.0-Asahi-ARCH

blazra, 2022-10-23:
Rgarding the GCC: It looks like a misunderstanding. The README says: "To build on a native arm64 machine, use make ARCH=.". You should add just `ARCH=` not `ARCH=arm64`. `ARCH=` sets a prefix for GCC - if you are building natively you can use just `gcc` therefore no prefix.

Regarding Rust: I think the problem is that you don't have the rust-std component. `rustup target install aarch64-unknown-none-softfloat` should install it. And then `make ARCH= CHAINLOADING=1` should work.

SuperKenVery, 2022-10-24:
Thanks. 

I tried to run the rustup command for another time, and this time it installed the rust-std component. I have no idea why it didn’t install this component the first time I ran it. That was super strange…

I am totally new to rust…

closed 2022-10-24 by SuperKenVery