AsahiLinux/m1n1 #270: Error when installing Asahi (on Mac OS)

Galorlx, 2022-12-17:
```Enter a name for your OS
» OS name (Asahi Linux): Asahi Linux

Using macOS 12.3 for OS firmware

Downloading macOS OS package info...
- 

Creating new stub macOS named Asahi Linux
Failed to run process: diskutil addPartition disk0s2 apfs 'Asahi Linux' 2499805184
Output: b'The new partition to be added will be of a filesystem personality of APFS apfs APFS 7C3457EF-0000-11AA-AA11-00306543ECAC\nThe new partition will be added following disk0s2\nStarted partitioning on disk0\nError: -69519: The target disk is too small for this operation, or a gap is required in your partition map which is missing or too small, which is often caused by an attempt to grow a partition beyond the beginning of another partition or beyond the end of partition map usable space\n'
root        : ERROR    Process execution failed
Traceback (most recent call last):
  File "/private/tmp/asahi-install/main.py", line 1054, in <module>
    InstallerMain().main()
  File "/private/tmp/asahi-install/main.py", line 865, in main
    while self.main_loop():
  File "/private/tmp/asahi-install/main.py", line 1017, in main_loop
    return self.action_install_into_free(parts_free)
  File "/private/tmp/asahi-install/main.py", line 305, in action_install_into_free
    self.part = self.dutil.addPartition(free_part.name, "apfs", self.osins.name, STUB_SIZE)
  File "/private/tmp/asahi-install/diskutil.py", line 250, in addPartition
    self.action("addPartition", after, fs, label, size, verbose=True)
  File "/private/tmp/asahi-install/diskutil.py", line 34, in action
    p = subprocess.run(["diskutil"] + list(args), check=True,
  File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['diskutil', 'addPartition', 'disk0s2', 'apfs', 'Asahi Linux', '2499805184']' returned non-zero exit status 1.
If you need to file a bug report, please attach the log file:
  /private/tmp/asahi-install/installer.log
```

That's the full log please help.

Galorlx, 2022-12-17:
Here is the log file: 
[installer.log](https://github.com/AsahiLinux/m1n1/files/10252750/installer.log)

marcan, 2022-12-18:
This is an Apple diskutil bug that seems to happen when you have a spurious Apple_Boot partition. Try running `diskutil eraseVolume free free disk0s3` first.

Galorlx, 2022-12-18:
> This is an Apple diskutil bug that seems to happen when you have a spurious Apple_Boot partition. Try running `diskutil eraseVolume free free disk0s3` first.

Yeah that worked great thank you. Also when would you say internal speaker support would be available?

closed 2022-12-19 by marcan