AsahiLinux/m1n1 #287: Mac Mini M2 - 2023 Support

spikerguy, 2023-02-18:
Hello Team,

As per [Feature Support](https://github.com/AsahiLinux/docs/wiki/Feature-Support#m2-series-device-specific-support) Mac Mini M2 - 2023 does not have installer support.

I am looking to buy Mac Mini so I am confused if M2 will have a working installer or not.  

I can try to help with the support but I do not have clear understanding of what is missing to get M2 support added, if someone can share more details with me then I can look into it and try to add support.

Is there any way I can connect to IRC on discord or matrix? anyone maintain any such bridge ?

Thank you for all your work.

jannau, 2023-02-19:
None of the devices released in 2023 is supported yet. M2 Pro/Max support is still being worked on. The main issue for adding support for the M2 Mac mini is the missing support for any display outputs. Apple changed how the HDMI port works on the M2/M2 Pro Mac minis. This breaks initializing the HDMI port and display in m1n1. I made an attempt to adapt it but it unfortunately doesn't appear to work. I moved on to support it in Linux first since that's easier to compare to macOS and more or less will do the same thing as macOS.
The same issue is expected to exist on the M2 Pro Mac mini.
This issue is unfortunately not an easy one to start with.

Adding support in the installer itself is not complicated, see AsahiLinux/asahi-installer#179. That is not ready for users as it misses support for the devices in m1n1 and the installer images.

I'm not aware of any discord/matrix IRC bridges. The main matrix bridge is unfortunately banned due to repeated abuse. You could try to use https://webchat.oftc.net/ and https://alx.sh/l/asahi-dev to catch up when you're not connected.

f-9t9it, 2023-02-19:
Can the device be used without hdmi output?  

I don't mind using uart and bringing it up on lan. 

I didnt find documentation on uart debugging or maybe i missed it.  

Can you point me to such documentation?  If i can use it without hdmi then i will order it else will order m1

jannau, 2023-02-19:
yes, uart and lan works.

relevant documentation is in https://github.com/AsahiLinux/docs/wiki/Tethered-Boot-Setup-(For-Developers) and https://github.com/AsahiLinux/docs/wiki/SW%3AHypervisor

Installation will be a little difficult.

Please stop using issues as support requests.

spikerguy, 2023-02-19:
Thanks for the links. 

I will go through them and also join in irc.

closed 2023-02-19 by spikerguy

KirMozor, 2023-07-22:
When will Asahi be available for Mac Mini M2?

nsdkin, 2023-07-22:
Is there any ETA for Mac Mini M2 support?

teras, 2023-07-24:
Hello team and first of all thank you for the hard work. For all the respect, I believe this should be an open issue. Asahi right now doesn't support Mac Mini M2. Even though this is not an easy job, and is work in progress, this issue IMHO should be open.

sampro1243, 2023-08-05:
ihave the same issue

jannau, 2023-08-05:
I have the issue that me too comments destroy the motivation to work on it. So I guess I won't today.

ivanhub, 2023-10-21:
Oh, how it will be astounding, when it will fully works on M2. Oh God, give developers supernatural power for that! :)