AsahiLinux/m1n1 #308: Custom Boot-Logo and Asahi Artwork folder

philmmanjaro, 2023-05-03:
When you get the tarball from the v1.2.6 release of m1n1 the folder **artwork** is empty.  `_artwork_commit_id=35626c31f5c5a8a884aaff93f864795026495742` is used within the asahi PKGBUILD. If I look at the Fedora Alternative I find this repo, which replaces the Asahi Logo with the one of Fedora: https://github.com/leifliddy/asahi-artwork

When I look at the SPEC file of **m1n1** I also see that Fedora replaces the logo: https://src.fedoraproject.org/rpms/m1n1/blob/rawhide/f/m1n1.spec#_64. Same goes for Nix if activated: https://github.com/tpwrules/nixos-apple-silicon/blob/main/apple-silicon-support/packages/m1n1/default.nix#L55-L67.

When compiling **m1n1** with an empty `artwork` folder it still compiles fine. So is the artwork folder still needed?

[image]

At boot selection you may see the Asahi Logo and the OS label based on the values given by the Asahi Installer. Which logo is used in that selection menu and how can it be replaced? Calling `update-m1n1` changes the boot logo but might not the logo in the selection menu.

Is there a plan to enable an easy way to rebrand so Distros can use the m1n1 bootloader but are able to use their own brand?

leifliddy, 2023-09-10:
So the artwork is used in three places: stage1, stage2, and bootpicker icon
If artwork was not specified when compiling m1n1 -- if would simply used the built-in artwork it already has: 
https://github.com/AsahiLinux/m1n1/tree/main/data (the bootlogo*.bin files)

https://www.reddit.com/r/AsahiLinux/comments/15zlil5/comment/jxtijfx/?utm_source=share&utm_medium=web2x&context=3

chadmed, 2025-04-13:
The macOS boot picker icon is an `.icns` file stored on the APFS stub volume, which is hidden by default. The boot logo inserted by m1n1 can be changed using 5291d5c

closed 2025-04-13 by chadmed