AsahiLinux/m1n1 #341: How to start Windows from m1n1

windowssub, 2023-10-15:
I have known the project named m1n1_windows, but its Readme file suggests how to install linux. I've also tried booting through grub on Asahi Linux, but it didn't work(jumped to macOS). I'm confused.

jannau, 2023-10-16:
You can't. Windows is out of scope for AsahiLinux.

Look at https://github.com/AppleWOA/m1n1_windows but as far as I'm aware it is WiP and updating README.md is low on the list of priorities so you get the unmodified upstream version.

BokBoy, 2023-10-19:
@jannau I got the bin file but asahi bootloader doesn't seem to like to boot into it. Can you help me, there isn't a issues in the M1N1 Windows.

Nissoco23, 2023-12-23:
@BokBoy did you have any luck getting it to boot?

BokBoy, 2023-12-23:
> @BokBoy did you have any luck getting it to boot?

Nup 😢

jailbreakmerebooted, 2024-05-03:
windows sucks, the solution is to use linux

BokBoy, 2024-05-03:
> windows sucks, the solution is to use linux

Games: WHAT ABOUT ME??!!
Yea I know what you mean when you say windows sucks though. I personally do believe any Unix based systems are better.

jailbreakmerebooted, 2024-05-04:
It's all about free software. Why paying for software you don't really own. In Linux case the world owns it. Everyone can build their perfect own Linux distro. Games are so overrated.. also, wine exists.

Games should also be open source and owned by the world.

I ❤️ OSS

BokBoy, 2024-05-05:
> It's all about free software. Why paying for software you don't really own. In Linux case the world owns it. Everyone can build their perfect own Linux distro. Games are so overrated.. also, wine exists.
> 
> Games should also be open source and owned by the world.
> 
> I ❤️ OSS

We've gone hell out of topic...

I agree with you mate. It's a pain when devs say it's 'freeware' then they ask you to pay to do like 90% of the items. Basically scamming you, especially when it's like parallels desktop or stuff like that; there's no good alternative. 

Yea I love the open source stuff.