AsahiLinux/m1n1 #342: No display over HDMI-out on M2 Mac Mini (j473ap)

3405691582, 2023-10-17:
Installed `UEFI environment only (m1n1 + U-Boot + ESP)` with expert mode. Display remains asleep and no video is displayed over HDMI-out.

`proxyclient/tools/shell.py` full logs from tethered boot follows:
```
$ ./proxyclient/tools/shell.py 
TTY> CPU init (MIDR: 0x611f0320)...
TTY>   CPU part: 0x32 rev: 0x10
TTY>   CPU: M2 Blizzard
TTY> 
TTY> boot_args at 0x803f0c088
TTY>   revision:     2
TTY>   version:      2
TTY>   virt_base:    0xffffffffff11c000
TTY>   phys_base:    0x80111c000
TTY>   mem_size:     0x1e0f00000
TTY>   top_of_kdata: 0x803f10000
TTY>   video:
TTY>     base:       0x9e5064000
TTY>     display:    0x0
TTY>     stride:     0xa00
TTY>     width:      640
TTY>     height:     1136
TTY>     depth:      32bpp
TTY>     density:    1
TTY>   machine_type: 0
TTY>   devtree:      0x1718000
TTY>   devtree_size: 0x5c000
TTY>   cmdline:      -v
TTY>   boot_flags:   0x0
TTY>   mem_size_act: 0x200000000
TTY> 
TTY> 
TTY> 
TTY> m1n1 v1.4.3
TTY> Copyright The Asahi Linux Contributors
TTY> Licensed under the MIT license
TTY> 
TTY> Running in EL2
TTY> 
TTY> Device info:
TTY>   Model: Mac14,3
TTY>   Target: J473
TTY>   Board-ID: 0x24
TTY>   Chip-ID: 0x8112
TTY> 
TTY> OS FW version: 13.5 (iBoot-8422.141.2)
TTY> System FW version: unknown (iBoot-10151.1.1)
TTY> Heap base: 0x803f10000
TTY> MCC: Initializing T8103 MCC...
TTY> MCC: Initialized T8103 MCC (8 channels)
TTY> MMU: Initializing...
TTY> MMU: RAM base: 0x800000000
TTY> MMU: Top of normal RAM: 0x9e201c000
TTY> MMU: Unmapping TZ0 region at 0x9e201c000..0x9e3d64000
TTY> MMU: TZ1 region has bad bounds 0x800000..0x800000 (iBoot bug?)
TTY> MMU: Unmapping TZ2 region at 0x9e6e08000..0x9ffe08000
TTY> MMU: Adding Device-nGnRE mapping at 0x400000000 (0x80000000)
TTY> MMU: Adding Device-nGnRE mapping at 0x480000000 (0x80000000)
TTY> MMU: Adding Device-nGnRE mapping at 0x580000000 (0x80000000)
TTY> MMU: Adding Device-nGnRE mapping at 0x600000000 (0x80000000)
TTY> MMU: Adding Device-nGnRE mapping at 0x6a0000000 (0x20000000)
TTY> MMU: Adding Device-nGnRE mapping at 0x6c0000000 (0x40000000)
TTY> MMU: Adding Normal-NC mapping at 0x9fff3c000 (0x24000)
TTY> MMU: Adding Normal-NC mapping at 0x9fff0c000 (0xc000)
TTY> MMU: Adding Normal-NC mapping at 0x9ffec4000 (0x24000)
TTY> MMU: Adding Normal-NC mapping at 0x9ffe94000 (0xc000)
TTY> MMU: Adding Normal-NC mapping at 0x9fffb8000 (0x4000)
TTY> MMU: Adding Normal-NC mapping at 0x9fff74000 (0x4000)
TTY> MMU: SCTLR_EL1: 30100180 -> 30901085
TTY> MMU: running with MMU and caches enabled!
TTY> AIC: Version 2 @ 0x23b0c0000
TTY> AIC: AIC2 with 1/8 dies, 1152/4096 IRQs, reg_size:08004 die_stride:04a00
TTY> WDT registers @ 0x23d2b0000
TTY> WDT disabled
TTY> pmgr: Cleaning up device states...
TTY> pmgr: Enabling 0.ANS, parent of active device APCIE_ST
TTY> pmgr: initialized, 303 devices on 1 dies found.
TTY> display: Display is external
TTY> display: Dummy framebuffer found, initializing display
TTY> dart: dart /arm-io/dart-dcp at 0x23130c000 is a t8110 (locked)
TTY> dart: allocating L2 tbl at 0, 1025 to 0x9fff20000
TTY> dart: dart /arm-io/dart-disp0 at 0x231304000 is a t8110 (locked)
TTY> dart: dart /arm-io/dart-disp0.0.0 L1 0 is real-time at 0x9fff3c000
TTY> dart: allocating L2 tbl at 0, 1025 to 0x9fff44000
TTY> rtkit(dcp): booting with version 12
TTY> rtkit(dcp): pre-allocated buffer (ep 0x1, dva 0x800859000, phys 0x9e6bd5000)
TTY> rtkit(smc): booting with version 12
TTY> display: waited 4000 ms for display status
TTY> display: connected:1 timing_cnt:35 color_cnt:10
TTY> display: timing mode: valid=1 1920x1080 60.00 Hz
TTY> display: color mode: valid=1 colorimetry=1 eotf=1 encoding=1 bpp=32
TTY> display: swapped! (swap_id=0)
TTY> display: Modeset took 4425 ms
TTY> display: Sleeping DCP (external)
TTY> DCP: dcp_shutdown(sleep=true) is broken with dptx-port, quiesce instead
TTY> fb init: 1920x1080 (30) [s=1920] @0x9e5064000
TTY> fb console: max rows 63, max cols 104
TTY> fb: display logo
TTY> Initialization complete.
TTY> Boot policy: sip0 = 127
TTY> Bringing up USB for early debug...
TTY> dart: dart /arm-io/dart-usb0 at 0x382f80000 is a t8110
TTY> USB0: initialized at 0x804959030
TTY> dart: dart /arm-io/dart-usb1 at 0x502f80000 is a t8110
TTY> USB1: initialized at 0x804959360
TTY> Waiting for proxy connection... . Connected!
Fetching ADT (0x0005C000 bytes)...
m1n1 base: 0x8037cc000
Have fun!
>>> 
```

jannau, 2023-10-17:
According to the log everything work as expected:

> TTY> display: connected:1 timing_cnt:35 color_cnt:10
> TTY> display: timing mode: valid=1 1920x1080 60.00 Hz
> TTY> display: color mode: valid=1 colorimetry=1 eotf=1 encoding=1 bpp=32

dcp has a connection to the display and will use 1 1920x1080 60.00 Hz as mode

TTY> display: swapped! (swap_id=0)
TTY> display: Modeset took 4425 ms
TTY> display: Sleeping DCP (external)
TTY> DCP: dcp_shutdown(sleep=true) is broken with dptx-port, quiesce instead
TTY> fb init: 1920x1080 (30) [s=1920] @0x9e5064000

m1n1 has set the framebuffer and reconfigured internally for 1920x1080 display.

What display is this? Does it support 1920x1080 60Hz as mode on other devices? It acts as if nothing is connected?

3405691582, 2023-10-17:
It's a HP 22cwa. In macOS, this works fine and is reported as 1920x1080 at 60Hz in System Settings, it behaves normally when driven by other devices and graphics cards, and this is a mode it supports in its published specifications.

When the device boots for m1n1, the attached display remains in sleep mode and doesn't power up.

leio, 2023-10-24:
At least with earlier iterations of the M2 HDMI m1n1 patches, it needed both stage1 and stage2 to have the support to reliably turn it on. I still usually don't see picture from stage1 unless I did a warm reboot, though with a new monitor I'm not sure it's just not the monitor being slow to light up (haven't startup cycled much yet).

That is to ask for an additional data point - does it also not work if you chainload stage2 m1n1-1.4.3+ after a stage1 m1n1-1.4.3+?

leio, 2023-10-24:
`proxyclient/tools/chainload.py -r path/to/stage2/m1n1.bin`
provided you have the udev rules to get `/dev/m1n1` and co symlinks for a stable ttyACM link (sometimes it's ttyACM0, sometimes ttyACM1)

At one point of time in earlier iterations, I also chainloaded stage2 twice to get the picture to light up - back when I didn't have any HDMI out supporting code in stage1 yet.

niggowai, 2023-10-28:
What version of m1n1 are you using in the 2nd stage?
I experienced Problems when running ~pacman -Syu~ last monday. The server didn't came up after an update and I wasn't at home...
After saving the data I reinstalled Asahi with the patched installer I used before. Asahi came up without problems but when I dated up, after a reboot, the same problem again. After another reinstall, with excluding m1n1 from the update, the reboot worked!

Tomorrow I'll try to track down the problem and open an issue to get the problem fixed. 

You can try to use an older version, but then you may have to build m1n1 to support display on m2 Mac mini. If it helps, i can provide you the package I installed Asahi with.

3405691582, 2023-11-18:
Sorry for the delay. I tried to chainload an m1n1 stage2. By reading the wiki carefully, I assume that the same stage2 binary can be obtained from https://cdn.asahilinux.org/os/uefi-only-20231012-1.zip, essentially. Subsequently extracting at `../u` and trying `env M1N1DEVICE=/dev/ttyACM0 ./proxyclient/tools/chainload.py -r ../u/esp/m1n1/boot.bin` did not yield any better results; the connection drops (presumably as this stage2 is "release" and drops proxy mode, or something), and the monitor didn't wake. If there's a better stage2 payload I can use, if you upload the blob somewhere, I can try that, otherwise I have to try and construct one myself.

In the meantime, I'll see if I can try with an alternate monitor.

3405691582, 2023-11-18:
HP 23xw and LG 27GP850-B also exhibited these symptoms.

3405691582, 2024-03-06:
This is now working as of recent releases, and with the same hardware, I can see U-Boot on the screen. Closing.

closed 2024-03-06 by 3405691582