AsahiLinux/m1n1 #348: broken m1n1 on j473

niggowai, 2023-10-30:
Since commit 0097d51149b4e129abf6498d25920cbc2dd5f62a m1n1 don't work anymore on `j473` macs.

Last monday I ran `pacman -Syu` on my mac, after a reboot the system didnt came up anymore.
when connecting to a screen, I saw the  Asahi logo on a blank screen with some pixels in the upper left corner beeing green and blue...
I played around a bit, reinstalled asahi with the installer i used before (Offline install with a patched version of m1n1 since m1n1 hadnt the support for `j473`). Asahi came up but when updating again, m1n1 didn't work again. So I reinstalled again and updated with m1n1 excluded and the reboot succeded. after that, I played around with different versions of m1n1 which I compiled from source to track down the fault.
When reverting the commit mentioned above, m1n1 v1.4.3 works again.
Since im pretty new to c, i think i can't help with much more than testing...

jannau, 2023-10-30:
Please try if increasing the `500` in https://github.com/AsahiLinux/m1n1/blob/9a277fadc900cbfaca8f560380babf167fe5744f/src/afk.c#L725 fixes the issue as well. Should be safe to increase in large steps, for example by multiplying by 10.
I'll work on the proper fix mentioned in the top v1.4.3 commit.

niggowai, 2023-10-30:
Yes, setting to `5000` fixed the problem :+1:

niggowai, 2023-11-04:
With m1n1 v1.4.6 I got the issue fixed
321a80c

closed 2023-11-04 by niggowai