AsahiLinux/m1n1 #350: How to resize partition size back?

Kotov584, 2023-11-02:


Capta1nT0ad, 2023-11-03:
Make sure to read what the operation you are selecting does before doing it- as the installer states:
`min   -  Shrink macOS as much as (safely) possible`

This means that macOS will be shrunken to the smallest safe size that allows error-free updates (it is around 40 GB + whatever files you have saved) and Asahi Linux will take the rest of the available space.

If Asahi Linux was allocated 300 GB, I have no idea what your disk size is but for example if I have a 512 GB SSD and want to allocate 30 GB to Asahi Linux:
`>> New size (50%): 482GB`

> I used only 150GB for Asahi Linux.

I am not sure what you mean: it allocated 300GB, _you_ 'used' only 150GB (??) and you want 30GB to ALX?

As for resizing, this is possible but you may have an easier time installing and reinstalling. See this wiki page:
https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet
_And **please** don't use that script..._

Kotov584, 2023-11-07:
Okay, let me clarify a few things here about what I did during installation. So I followed all steps by steps when it asked for a new size. I selected New Size (Min), but after I did the resizing, I noticed it took up almost all of my drive space. I have only a 512GB drive, and it took more than 300GB. When I noticed it after it did the resizing, I was a little bit shocked and confused. Like, why would it take up the entire space if I asked it to use the minimal space possible? I probably should have specified GB by myself if I knew this before. So as a result, I finished up installing 150GB for Asahi Linux, and macOS does not see that I have yet another like 150GB as it is reformatted to Linux format. Now I have 7GB free on my Mac because the Asahi Linux resizing process was done wrong. Otherwise, installation went smoothly but took some time due to the large 300GB file size.

Capta1nT0ad, 2023-11-07:
> Like, why would it take up the entire space if I asked it to use the minimal space possible? 

Because, as the installer states, it resizes **macOS** to the smallest size possible, not Asahi Linux.

> Now I have 7GB free on my Mac

This shouldn't happen, there should be far more space left than that- were you using expert mode?(?!?!)

> Otherwise, installation went smoothly but took some time due to the large 300GB file size.

So is there a problem?

From what I am getting at, your drive looks like this:

*(simplified to conform with traditional Linux drive partitioning, system/EFI partitions not included, not to scale)*

| macOS | Asahi Linux | Free Space |
| -------- | -------- | -------- |
| ~200GB | 150GB | 150GB | 

Is this correct?

Kotov584, 2023-11-07:
Yeah, but macOS have only 186GB capacity now. I guess, resizing partitions instructions should bee more clear. Or I missed some part.