AsahiLinux/m1n1 #358: Is there a way to pass Linux cmdline from m1n1?

fathonix, 2023-11-23:
I don't really need U-Boot and GRUB so I built m1n1 with the kernel, dtb and initramfs, but I haven't found a way to pass kernel cmdline so I have to mount the root manually in initramfs at boot. I have set default values in Debian initramfs-tools conf but for some reason busybox mount detects it as a syntax error. I saw m1n1 can read cmdline, but it's only used for debugging and chainloading I think. I don't want to pull and rebuild the kernel only to change the cmdline though.

dkwo, 2024-04-25:
what is described in m1n1 user guide at 'Configuring for stage 2' works for me.
this way i can e.g. change the kernel loglevel etc.
it should also work in the way it's done in the update-m1n1 script.
both of these seem to be bash-specific though, as they were not working for me with just /bin/sh

fathonix, 2024-04-25:
This is exactly what I'm looking for, thanks!

closed 2024-04-25 by fathonix