AsahiLinux/m1n1 #435: Can't boot asahi fedora after 1.4.19-1 update

rlippmann, 2025-02-01:
Fedora offered an update to m1n1, and I stupidly said yes.

Now asahi goes into a boot loop.  Just loads the splash screen and restarts.

rlippmann, 2025-02-01:
Wiped and did a fresh install.  Rebooted fine.  But doing a dnf refresh/upgrade, same problem.

jannau, 2025-02-01:
this is fixed via https://github.com/AsahiLinux/m1n1/pull/434 which is being prepared as update in https://bodhi.fedoraproject.org/updates/FEDORA-2025-3b3113923c

This issue affects M2 Pro/Max Macbook Pros.

In the case this issue is hit thew system can be recovered from Macos or Macos recovery by:
```
mkdir /Volumes/EFI
mount -t msdos /dev/disk0s4 /Volumes/EFI
cp /Volumes/EFI/m1n1/boot.bin.old /Volumes/EFI/m1n1/boot.bin
```
This assumes a standard install with a single Macos partition and a single installation using m1n1.

Keeping this issue open until the fixed version is available.

rlippmann, 2025-02-01:
Yes, I have a 16" M2 Max Macbook Pros.

Thanks for the quick response for the fix!

closed 2025-03-29 by chadmed