AsahiLinux/m1n1 #456: root account locked on boot using a specific monitor (MSI portable)

davidberglund, 2025-03-29:
On a fresh Asahi install, when booting Asahi with my MSI Optix MAG162VDE Portable Monitor (connected with HDMI and USB-C for power) it does not boot up to the login manager but instead says that the root account is locked. When this happened the first time I thought that - while strange - I had probably accidently pushed some keys or whatever (rendering my asahi install useless). But after reinstall of Asahi and first booting with my regular monitor and verifying that the install was working, I rebooted with my MSI portable monitor again which got me into the same state with a locked root account.

Is this report useful at all? Can I get out of this state somehow? And can I add more info to make this issue more useful?

Thanks

davidberglund, 2025-03-31:
Oh, wait, Asahi boots up without issue if I powered the monitor from a different USB source other than the Mac Studio USB-C. I know that the Thunderbolt/USB ports are not yet supported on this machine so is it the case that even using these only power is not supported? At least I know how to use the Asahi with my portable monitor now.

vulpes2, 2025-04-08:
What you're seeing is the dracut emergency shell, caused by all the failed mount targets when the internal nvme drive wasn't being detected. Have seen this a few times myself, a reboot seems to bring it back, and it happens occasionally even when no USB devices are connected.